ansys/pyaedt #231: parameter  solution_type versus solutiontype

PipKat, 2021-07-07:
In the FieldAnalysis3D class (pyaedt/application/Analysis3D),  solutiontype (with no hyphen) is used. In the FieldAnalysis2D class (Ansysis2D in the same folder),  (and in most other places, solution_type is used.

akaszynski, 2021-07-23:
@MaxJPRey, please submit a PR refactoring `solutiontype --> solution_type`.  Would be great to have more snakecase in our API.  See:
https://github.com/pyansys/PyAEDT/blob/efe8d219be974fa8a164d84ca9bc5c0e1b32256c/pyaedt/application/Analysis3D.py#L56-L59

MaxJPRey, 2021-07-26:
@akaszynski I fixed that in an upcoming pull request. However, I can see that `projectname` and `designname` keyword argument are not snake_case. Should I create 2 new issues and fix them in 2 new separate PR?

PR #340 (2021-07-27): Rename keyword argument solutiontype into solution_type in order to b…
Fix #231 issue.
Rename keyword argument `solutiontype `into `solution_type `in order to be snake_case.
closed 2021-07-27 by MaxJPRey