ansys/pyaedt #269: Let's avoid adding print statements and instead use the standard python logging module.  Users won't want their stdout polluted.

voss70, 2021-07-13:
Let's avoid adding print statements and instead use the standard python logging module.  Users won't want their stdout polluted.

_Originally posted by @akaszynski in https://github.com/pyansys/PyAEDT/pull/264#discussion_r668847453_

akaszynski, 2021-07-14:
What you'll want to do is define a module level logger at the top of each module and then log using that.  For example:

```py
import logging
log = logging.getLogger(__name__)
log.setLevel('CRITICAL')
```

Then have the messenger print to that logging module.  You can change the logging level of this module via
```
logging.getLogger('pyaedt.<module>').setLevel(logging.INFO)
```

Alternatively, you could create a logger just for the messaging module and set its level there.  Regardless, we need to convert all calls to print to use a logging module and then have the level of that logger set based on the logging level requested by the user.

akaszynski, 2021-09-03:
Pinging @MaxJPRey to take a look at this as well since it's forcing us to skip certain doctests.