ansys/pyaedt #271: Provide update method for Design Class

voss70, 2021-07-14:
At present, the following functions call the Design.__init__() function due to the lack of an appropriate update function.

insert_design
set_active_design
copy_design_from
duplicate_design

It is proposed to develop an update() function to avoid losing the pointer to Design due to the constructor call

voss70, 2021-07-24:
In this context we should be able to update the primitives property of the modeler whenever the design is updated. This will enable us to eliminate the modeler code:

@property
    def primitives(self):
        """ """
        if self._primitivesDes != self._parent.project_name + self._parent.design_name:
            self._primitives.refresh()
            self._primitivesDes = self._parent.project_name + self._parent.design_name
        return self._primitives.

voss70, 2021-07-24:
From there we might consider doing the right thing and inheriting the Primitives classes into Modeler directly. That would allow the user to write aedtapp.modeler.create_box directly and eliminate a bunch of reduntant properties like oproject etc, with the primitives classes in the background. We could keep an aedtapp.modeler.primitives pointer for backwards compatibiility, i.e. self.primitives = self.  One to think about during the summer vacation ...

Samuelopez-ansys, 2022-08-30:
@maxcapodi78 I think this issue could be closed right? Design class has changed a lot and now these methods are independent

closed 2022-08-31 by maxcapodi78