ansys/pyaedt #272: JobManager.py in Application folder

PipKat, 2021-07-14:
def update_config_name(self, name, machinename):
    """Update the name of the machine.
    
I believe there is an error here (line 144) as this is supposed to be for updating the name of the configuration file yet asks for the name of the machine. There was already a method for updating the machine name.

Also, I don't  see this page appearing yet in the API Doc.

akaszynski, 2021-07-14:
> Also, I don't see this page appearing yet in the API Doc.

Is that on dev.aedtdocs.pyansys.com?  If so, that will have to be fixed after #176 is merged.

Also, when possible, use the github permalink to provide exact code reference.  Clicking on the line numbers on the online file viewer (holding shift and clicking the bottom of the range if you want the range), then clicking on the three dots to the left let's you generate a permalink to the documentation:
[image]

https://github.com/pyansys/PyAEDT/blob/ac0c21216dbb86a9e3c004d156d8b02dc1e25981/pyaedt/application/JobManager.py#L120-L143

akaszynski, 2021-07-23:
Looked at this again.  @maxcapodi78, are we even using this?
https://github.com/pyansys/PyAEDT/blob/1b7e2216be33d7ee6f2298a00f64f98730185a6f/pyaedt/application/JobManager.py#L140-L166

closed 2021-11-04 by MaxJPRey