ansys/pyaedt #333: analyse versus analyze

PipKat, 2021-07-23:
In hfss.py (and likely other files), some methods use analyse_ and others use analyze_. We should always use analyze as we are to use American spellings. Also, in hfss.py, we are missing an "in" in assign in this method: assig_voltage_source_to_sheet

akaszynski, 2021-07-23:
@MaxJPRey, please rename `assig_voltage_source_to_sheet` to `assign_voltage_source_to_sheet`:
https://github.com/pyansys/PyAEDT/blob/130ecc7f80c648a7d3eb2606e8c9d8ce9e1b89ed/pyaedt/hfss.py#L1712

Additionally, create a separate PR with an `analyse --> analyze` refactor.

maxcapodi78, 2021-07-26:
I agree. Please make sure that Examples are corrected accordingly

MaxJPRey, 2021-07-26:
@maxcapodi78 Sure, I will check the examples.

MaxJPRey, 2021-07-26:
@akaszynski I created a separate issue regarding the renaming of `assig_voltage_source_to_sheet `to `assign_voltage_source_to_sheet`.
See #342

PR #347 (2021-07-28): Replace 'analyse' by 'analyze'.
Fix #333.
Replace 'analyse' by 'analyze' in pyAEDT.
closed 2021-07-28 by MaxJPRey