ansys/pyaedt #351: UDP Parametrization needs expanding to handle Version and NoOfParameters fields

voss70, 2021-07-27:
This is a recorded version of drawing an RMxprt UDP. In our pyaedt implementation of primitives.create_udp we do not supply the parameters Version of NoOfParameters. This  causes the RMXprt/PMCore upd to fail. 
oEditor.CreateUserDefinedPart(
	[
		"NAME:UserDefinedPrimitiveParameters",
		"DllName:="		, "RMxprt/PMCore.dll",
		"Version:="		, "12.0",
		"NoOfParameters:="	, 13,
		"Library:="		, "syslib",
		[
			"NAME:ParamVector",
			[
				"NAME:Pair",
				"Name:="		, "DiaGap",
				"Value:="		, "RotorDia"
			],

tizianrot, 2022-02-03:
Dear All,

I just came across the same issue. When calling a rmxprt UDP from Maxwell 3D with parameters different from the default ones the following happens:

- Warning message: User Defined Primitive: Function MapParametersDefinitionVersions not found in RMxprt/TransCoil.dll. Using default parameters. (2:04:10 pm  Feb 03, 2022)
- the parameter is not taken
- the default scalar is used instead
- this happens assigning both scalar values and desin parameters
- it does not happen with IronPython/desktop scripting, so it is somewhere in pyaedt

Tried with TransCoil and TransCore

Thank you,

Tiziana

maxcapodi78, 2022-05-04:
Fixed

closed 2022-05-04 by maxcapodi78