ansys/pyaedt #364: Add doc coverage CI/CD

akaszynski, 2021-07-30:
Add CI/CD unit test discovery/coverage.

1.  Loops through all modules in the ``pyaedt`` directory.  Record if classes, modules, and functions have a doctest example section.  Skip private methods.
2.  Produce a text report indicating module coverage.  For example:
3. 
```
Name                                         Methods   Miss  Cover
------------------------------------------------------------------
pyaedt/hfss.py                                    53     16    70%
pyaedt/<other_module>.py                         200    200     0%
```

3. Add this to CI/CD in github within the doctest in ``ci-build.yml`` as a "always run" step.

MaxJPRey, 2021-08-24:
This will be covered in one of the tag/label created under "Projects".

closed 2021-08-24 by MaxJPRey

akaszynski, 2021-08-24:
> This will be covered in one of the tag/label created under "Projects".

Added.