ansys/pyaedt #370: Maxwell 2D example in Example gallery does not work in PyCharm

tizianrot, 2021-08-05:
Hallo,

some functions in this example gave me errors, i.e:
1) function: duplicate_along_line is no longer an attribute of rect1 as in the example: 
added = rect1.duplicate_along_line([14,0,0])
2) edge list no longer exists as attribute as in the example: 
m2d.assign_balloon(region.edges)
3) the function: 
m2d.save_project(pName+'.aedt') 
did not save

below you can find how I corrected the script.
Thank you
Tiziana
**************************************
**************************************
import os
from pyaedt import Maxwell2d

NonGraphical = False

pathN = "C:\\MyFolder\\"
pName = "Example"
dName = "MyMaxwell2DDesign1"
sName = "MySetupAuto"
sType = "TransientXY"

m2d=Maxwell2d(projectname=pathN+pName,designname=dName,solution_type="TransientXY", specified_version="2021.1",NG=False)

rect1 = m2d.modeler.primitives.create_rectangle([0,0,0],[20,10],"Rectangle1", "copper")
m2d.modeler.duplicate_along_line("Rectangle1",[14,0,0],nclones=2,attachObject=False)
region = m2d.modeler.primitives.create_region([100,100,100,100,100,100])

region_edge_list = m2d.modeler.primitives.get_object_edges("Region")
m2d.assign_balloon(region_edge_list,bound_name= "MyBalloon")

m2d.assign_winding(["Rectangle1","Rectangle1_1"],winding_type="Current",current_value=2, parallel_branches=1, name="PH_A")

setup = m2d.create_setup(setupname="MySetupAuto")
setup.props["StopTime"] ="0.02s"
setup.props["TimeStep"] = "0.0002s"
setup.props["SaveFieldsType"] = "Every N Steps"
setup.props["N Steps"] = "1"
setup.props["Steps From"] ="0s"
setup.props["Steps To"] = "0.002s"
setup.update()

m2d.post.create_rectangular_plot("InputCurrent(PH_A)",primary_sweep_variable="Time", families_dict={"Time":["All"]}, plotname="Winding Plot 1")
m2d.analyze_setup("MySetupAuto")

akaszynski, 2021-08-07:
Thanks posting the fix!  @MaxJPRey, while @maxcapodi78 is out on vacation, could you handle patching this?

MaxJPRey, 2021-08-09:
> Thanks posting the fix! @MaxJPRey, while @maxcapodi78 is out on vacation, could you handle patching this?

@akaszynski . Sure I will.

MaxJPRey, 2021-08-09:
Hi @tizianrot ,
Looking at the code history, the example you mentioned (Maxwell2D_Transient) has been updated 26 days ago in the commit #264. In the same commit, a lot of modifications were made to the `Object3D `class in order to modify the returned type.
I have tried the example with the current release and is is working fine on my machine.

I am assuming that you are running the latest version of the example but with an old version of the pyAedt package.
Consequently, you had to use the code you created and suggested in this issue above.

I hope it helps. Please let us know if you can run the example with the last version of the pyAedt package.

Thanks for your feedback, it is truly appreciated.

Maxime.

closed 2021-09-02 by MaxJPRey