ansys/pyaedt #382: Add syntax to remove variables

dcrawforAtAnsys, 2021-08-16:
Please make it possible to remove a design or project variable using the following syntax:
>>> from pyaedt import Hfss
>>> app = Hfss()
...
>>> app['var'] = None

This would remove 'var1' if it is not used.

akaszynski, 2021-08-16:
Would it be better to use:
```py
>>> from pyaedt import Hfss
>>> app = Hfss()
...
>>> del app['var']
```

This way it's more like a `dict`.

dcrawforAtAnsys, 2021-08-17:
Yes. This is better.

akaszynski, 2021-08-17:
@maxcapodi78, @MaxJPRey 

Let's implement the following `__delitem__`:
```py
def __del__(self, key):
    """Implement del with array name or index."""
    del self.variable_manager(key)
```

We should also consider implementing a nested interface via a `variables` property:

```
    @property
    def variables(self):
        """Variables.."""
        self.variable_manager
```

While "flat is better than nested", in this case I think this is too flat.  Application should act as an "application", not as a variables dictionary as well.  This is something we'd deprecated in the next minor release.

closed 2021-09-02 by MaxJPRey