ansys/pyaedt #384: Unable to instantiate app object if desktop has been instantiated

dcrawforAtAnsys, 2021-08-18:
>>> import os
>>> import pyaedt
>>> aedt_version = '2021.2'
>>> projectname = 'MicroDoppler_with_ADP'
>>> designname = 'design1'
>>> project_path = r'C:\Users\dcrawfor\Documents\Ansoft\applications\microdoppler_adp'
>>> oDesktop = pyaedt.Desktop(specified_version=aedt_version,
>>>                          AlwaysNew=False,
>>>                          release_on_exit=True)
----
Launching AEDT installation C:\Program Files\AnsysEM\AnsysEM21.2\Win64
===================================================================================
pyaedt Info: pyaedt v0.3.8
pyaedt Info: Python version 3.7.9 (heads/bdighole/python/3.7.9-dirty:eff3adb, Mar 23 2021, 08:16:15) [MSC v.1910 64 bit (AMD64)]
-----------
>>> app = pyaedt.Hfss(projectname = os.path.join(project_path, projectname),
>>>                   designname = designname,
>>>                   specified_version = aedt_version, 
>>>                   solution_type = 'SBR+', 
>>>                   AlwaysNew = False, release_on_exit = True)
---
c:\users\dcrawfor\appdata\roaming\aedt_automation_dev\lib\site-packages\pyaedt\application\Design.py in oproject(self, proj_name)
    994             self._oproject = self._desktop.GetActiveProject()
    995         else:
--> 996             if proj_name in self._desktop.GetProjectList():
    997                 self._oproject = self._desktop.SetActiveProject(proj_name)
    998             elif os.path.exists(proj_name):

AttributeError: 'Desktop' object has no attribute 'GetProjectList'

akaszynski, 2021-08-19:
@ring630, could you please take a look at this?

ring630, 2021-08-20:
@akaszynski ,sure.

ring630, 2021-08-20:
It looks like  app = pyaedt.Hfss is trying to instantiate a new AEDT desktop, which is not allowed. Use **With** sentence can avoid double instantiation. There can be an enhancement to allow the second AEDT desktop to be created.

`import os
import pyaedt

aedt_version = '2021.2'
projectname = 'Project27'
designname = 'HFSSDesign1'
project_path = ""

with pyaedt.Desktop(specified_version=aedt_version,
                    AlwaysNew=False,
                    release_on_exit=True):

    app = pyaedt.Hfss(projectname=os.path.join(project_path, projectname),
                      designname=designname,
                      specified_version=aedt_version,
                      solution_type='SBR+',
                      AlwaysNew=False, release_on_exit=True)`

akaszynski, 2021-08-20:
This is related to #259.  There needs to be a refactor of how we create the desktop.  Also see #388.

maxcapodi78, 2021-08-25:
@akaszynski @ring630 @dcrawforAtAnsys the error is even much simpler. you cannot initialize pyaedt.Desktop in oDesktop Variable. oDesktop is a global variable used by the com object
if you do simply
d = pyaedt.Desktop(specified_version=aedt_version, AlwaysNew=False,release_on_exit=True)
the script will work

closed 2021-08-25 by maxcapodi78