ansys/pyaedt #385: No method is accessible in class edb.Cell

ring630, 2021-08-18:
Reproduce the problem by below code
```
import clr

clr.AddReference('System.Xml')
from pyaedt.edb import Edb

aedb_path = "Galileo.aedb"
edb = Edb(edbpath=aedb_path, edbversion="2021.1")
edb.edb.Cell.Create()
```

```
Traceback (most recent call last):
  File "C:/Users/hzhou/AppData/Roaming/JetBrains/PyCharmCE2020.1/scratches/json_test.py", line 10, in <module>
    edb.edb.Cell.Create()
AttributeError: Create
```

maxcapodi78, 2021-08-23:
I would suggest to try using:
 edb.edb.Cell.Create(edb.db,  edb.edb.Cell.CellType.CircuitCell, "MyNewCell")

ring630, 2021-08-23:
> I would suggest to try using:
> edb.edb.Cell.Create(edb.db, edb.edb.Cell.CellType.CircuitCell, "MyNewCell")

I still get this error

Traceback (most recent call last):
  File "C:/Users/hzhou/AppData/Roaming/JetBrains/PyCharmCE2020.1/scratches/scratch.py", line 8, in <module>
    edb.edb.Cell.Create(edb.db, edb.edb.Cell.CellType.CircuitCell, "MyNewCell")
AttributeError: Create

maxcapodi78, 2021-08-23:
it is because you are missing one level
the right command should be (with two Cell levels)

edb.edb.Cell.Cell.Create(edb.db, edb.edb.Cell.CellType.CircuitCell, "MyNewCell")

maxcapodi78, 2021-08-23:
Anyway we may add a create_cell method in Edb Class

maxcapodi78, 2021-08-25:
@ring630 can you close this issue since my last comment worked for you?

closed 2021-08-25 by ring630