ansys/pyaedt #388: pyaedt.Hfss() creates new instance of identical project/design

dcrawforAtAnsys, 2021-08-20:
If app.Hfss() is called with an identical signature (i.e. same project name, path, design name, etc) and AlwaysNew=False, a new instance the exact same project is created.
>>> aedt_version = '2021.2'
>>> projectname = 'MicroDoppler_with_ADP'
>>> designname = 'design1'
>>> project_path = r'C:\Users\dcrawfor\Documents\Ansoft\applications\microdoppler_adp'

run this twice. I would expect the 2nd call to just return the same object and not create a new instance, since the project already exists.

>>> app = pyaedt.Hfss(projectname = os.path.join(project_path, projectname),
>>>                   designname = designname,
>>>                   specified_version = aedt_version, 
>>>                   solution_type = 'SBR+', 
 >>>                  AlwaysNew = False, release_on_exit = True)

akaszynski, 2021-08-20:
This is related to #259. There needs to be a refactor of how we create the desktop. Also see #384.

maxcapodi78, 2021-08-23:
Fixed

closed 2021-08-23 by maxcapodi78

akaszynski, 2021-08-23:
Please post the PR that resolved this issue.  Let's also link issues with PRs so that when they're resolved these issues will be linked and automatically closed.