ansys/pyaedt #429: Maxwell2D->Master/Slave

Javpyer, 2021-09-03:
How can I set the Master/Slave boundaries? I'm not finding corresponding API Document. Thanks!

Zwl20085, 2021-09-06:
> How can I set the Master/Slave boundaries in Maxwell2D? I'm not finding corresponding API Document. Thanks!

I think you can access the corresponding function through com object of AEDT (maxwell2d.oeditor or maxwell2d.odesign ). you can find the name of the function to set master/slave in <<maxwell scripting.pdf>>, which is stored in AEDT if you have installed in local help.

maxcapodi78, 2021-09-13:
It's not there yet. I'll put in the backlog of new feature

maxcapodi78, 2021-09-13:
@Zwl20085  you are right. any function not wrapped by pyaedt is accessible through the oeditor or odesign object.

maxcapodi78, 2021-09-13:
@Javpyer  assign_master_slave method is now in #453

closed 2021-11-05 by maxcapodi78