ansys/pyaedt #451: get_object_from_name() is well hidden.

dcrawforAtAnsys, 2021-09-11:
@maxcapodi78
I think the method 
  Hfss.modeler.primitives.get_object_from_name("object_name")
which returns the instance of Object3D is very useful, but is difficult to find navigating through the class heirarchy. I think we should bring this to a higher level. Maybe even as a method associated with the app?

dcrawforAtAnsys, 2021-09-11:
This is related to the return value from methods in Object3D. If the return value were the object instead of the name, then it would not be necessary to get the object using the name and call .get_object_from_name()

maxcapodi78, 2022-05-04:
You can access it quickly directly from modeler __getattr__, 
ex. 
hfss.modeler["Region"]

closed 2022-05-04 by maxcapodi78