ansys/pyaedt #452: AttributeError: module '__main__' has no attribute 'oDesktop'

zebanaqviWAT, 2021-09-12:
I am receiving this error on using 21R1. 
```
import os
import pyaedt
from pyaedt import Hfss
from pyaedt import Desktop

NG =True
d = Desktop("2021.1", NG=NG)
Launching AEDT installation C:\Program Files\AnsysEM\AnsysEM21.1\Win64
===================================================================================
Traceback (most recent call last):

  File "<ipython-input-11-d4d524c49518>", line 7, in <module>
    d = Desktop("2021.1", NG=NG)

  File "C:\Users\znaqvi\Anaconda3\lib\site-packages\pyaedt\desktop.py", line 504, in __init__
    self._main.oDesktop.RestoreWindow()

AttributeError: module '__main__' has no attribute 'oDesktop'
```
Passing "2021.2" has no issues.
```
import os
import pyaedt
from pyaedt import Hfss
from pyaedt import Desktop

NG =False
d = Desktop("2021.2", NG=NG)
Launching AEDT installation C:\Program Files\AnsysEM\AnsysEM21.2\Win64
===================================================================================
pyaedt Info: pyaedt v0.3.7
pyaedt Info: Python version 3.8.3 (default, Jul  2 2020, 17:30:36) [MSC v.1916 64 bit (AMD64)]
```
There is a thread, now closed, about the same error but different situation than mine.  I tried to use both NG= True/False with an already open AEDT window of 21R1/R2, closing all sessions and starting blank etc. Is there a recommended way to start a session of desired release?

maxcapodi78, 2021-09-13:
@zebanaqviWAT  Hello. I'm not able to reproduce the issue. in my machine I'm able to start both versions of AEDT without issues. Are you sure that 2021.1 env variable are correctly setup? Have you rebooted machine after installing 2021.1AEDT?

maxcapodi78, 2021-09-13:
@zebanaqviWAT another suggestion: Initializing using Desktop is not useful if you are gonna use only 1 application. Instead you can do directly:
`hfss = Hfss(specified_version="2021.1", NG=False, AlwaysNew=False)`

that command will initialize a desktop and start Hfss on it.

zebanaqviWAT, 2021-09-20:
[image]
Massimo, 
Thanks for your help.
- I see these in my Environment Variables - how to correctly set them up? 
- If correct, I have surely rebooted the machine several time in the past since downloading 21R1. (I have 22R1 as well).
-  The other suggestion works perfectly, launching any version I want. However how do I elegantly close the AEDT? Says  
[image]

MaxJPRey, 2021-12-02:
Hi @zebanaqviWAT,
Do you still face any remaining issues regarding AEDT?

zebanaqviWAT, 2021-12-02:
Hi @MaxJPRey , I think we can close this as eventually I forced the AEDT to close, as I wasn't sure how to close it 'elegantly'.

MaxJPRey, 2021-12-02:
Thanks @zebanaqviWAT for the feedback.

closed 2021-12-02 by MaxJPRey