ansys/pyaedt #489: move_edge_along_normal()

dcrawforAtAnsys, 2021-09-23:
Method needs to be added to FacePrimitive in Modeler.Object3D.py

maxcapodi78, 2021-11-15:
@dcrawforAtAnsys Do you mean move_faces_along_normal? I don't think you can move an edge that is connected to a faces without detaching it

dcrawforAtAnsys, 2021-12-10:
In this case I wanted to move the edge of a rectangle in the normal direction. Move edge along normal is valid for a straight edge in a planar 2d object. I often use the following commands to create a port on the end of a trace (rectangle) with the following steps:

1. Select face
2. Create object from face
3. Select top/bottom edges
4. Move edge along normal (to make the connection to the top/bottom ground planes, for example in stripline)
5. Select the lateral edges
6. Move edge along normal (to make sure the port is wide enough to avoid near-field coupling between the edge pec and ground.
This ensures that the port retains the parameterization of the trace. So if the end of the trace moves because of a parameter change, the port surface will move with it.

maxcapodi78, 2022-05-04:
Fixed in #1134

closed 2022-05-04 by maxcapodi78