ansys/pyaedt #517: is there any recommended methods for use pyansys on linux platform?

alien008, 2021-10-22:
i am trying to use a student version of ansys electronics on linux , I want to use the pyansys package in pycharm to control ansys electronics, but I need to download pythonnet to use pyansys, and to download mono to use pythonnet. but I always encounter "UserWarning: Pythonnet is needed to run pyaedt within Linux." when I run the case code. i looked up on the web and some said the mono version my be not proper, but i tried several time , then the new errors come about :"Your mono runtime and class libraries are out of sync" Or "The assembly mscorlib.dll was not found or could not be loaded. It should have been installed in the `/usr/lib/mono/4.5/mscorlib.dll' directory.".  i think may be the mono path was not configured properly, then i changed the mono path , a new error "The assembly mscorlib.dll was not found or could not be loaded. It should have been installed in the `/usr/lib/mono/4.5/mscorlib.dll' directory."  showed up. So, could anyone give me a guiding process of the setup . thank you!

maxcapodi78, 2021-11-05:
Actually you cannot use CPython in Linux but you have to use Ironpython. You can open AEDT and execute a script with Pyaedt. the RunScriptAndExit option can be used also. As an alternative you can use pyaedt in remote and have Jupyter or PyCharm on Windows machine. Please give a look to documentation

closed 2021-11-05 by maxcapodi78

alien008, 2021-11-06:
thank you so much !

akaszynski, 2021-11-06:
@maxcapodi78, let's add a section in the user guide detailing the use of pyaedt on Linux in IronPython, or Windows CPython.

amrit-poudel, 2021-12-27:
@maxcapodi78: Is it possible to use pyAEDT  package in Linux machine as a client (with no ANSYS installed on it) and Windows machine as a server with ANSYS installed on it ? Can pyAEDT use CPython in Linux in such a case? If not what is the recommended way to use pyAEDT package to launch ANSYS in a remote Windows machine? Thanks!