ansys/pyaedt #561: Clean Stop / Abort Command in HFSS

tungnguyen1003, 2021-11-10:
Would anyone employ the code to execute Clean Stop or Abort command? Appreciate your help.

maxcapodi78, 2021-11-10:
that would be though but I'll investigate.

kbhagat-ansys, 2021-11-22:
I wonder how it would work. 
Currently if someone gives `oDesign.Analyze()` then that function only goes to the next script command when its completed. So when will user be able to do an Abort? This change might require development to step in and essentially allow a argument like `oDesign.Analyze(dont_wait_for_simulation_to_finish=True)` 

@maxcapodi78

maxcapodi78, 2022-08-31:
this cannot implemented at this time since AEDT API doesn't support it

svandenb-dev, 2023-02-20:
closing this issue as not supported by the API.

closed 2023-02-20 by svandenb-dev