ansys/pyaedt #588: draw_circle message error

cristianbabetto, 2021-11-17:
Good morning,

In Maxwell2d transient, when I use the command
`circ = m2d.modeler.primitives.create_circle([0,0,0], 20)
`
a circle is drawn but I get the following error messages:
`
*Global - Messages
  [info] pyaedt v0.2.23
  [info] Python version 3.8.10 (tags/v3.8.10:3d8993a, May  3 2021, 11:48:03) [MSC v.1928 64 bit (AMD64)]
PSA_ID_711_v2_ANSYSEM_2D (D:/AC_WINDING_LOSS_STUDY/MAXWELL/)
  Motor-CAD PSA_ID_711_v2 (Transient, XY)
    [error] Script macro error: Invalid PropServer 'NewObject_ZNCOWB'. (11:30:12  nov 17, 2021)
    [error] Script macro error: Invalid PropServer 'NewObject_ZNCOWB'. (11:30:12  nov 17, 2021)
`

maxcapodi78, 2021-11-18:
@cristianbabetto Please update to latest version of pyaedt (0.4.8). That issue has been fixed from pyaedt 0.4.x

cristianbabetto, 2021-11-18:
Ok thanks. Now it works.

closed 2021-11-19 by maxcapodi78