ansys/pyaedt #604: Unit test Errors if Redirecting Desktop\ProjectDirectory while running Unit Tests

kbhagat-ansys, 2021-11-24:
If I redirect Desktop\ProjectDirectory through BatchOptions when running the unit tests 1 test gives an error (`test_25_copy_solid_bodies (test_98_Icepak.TestSequenceFunctionsGenerate)`). 
Sample Command when running Unit Tests:
```batch
"C:\Program Files\AnsysEM\AnsysEM21.2\Win64\ansysedt.exe" "-RunScriptAndExit" "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\_unittest_ironpython\run_unittests.py" "-LogFile" "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\AEDT_Message_Manager.log" "-Ng" "-features=SF6694_Non_Graphical_Command_Execution" "-BatchOptions" "tempdirectory='D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\temp' Desktop\ProjectDirectory='D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639' Desktop\PersonalLib='D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\PersonalLib' Desktop\userlib='D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\userlib'"
```

Here is the Error I get from the Message Manager log: 
`[error] The project file path "D:ARMPyAEDT_UnitTests_IronPython_11242021151639\IcepakCopiedProject.aedt" is invalid. The project file path cannot be empty and the directory must already exist. The characters \/:*?"<>| cannot be in the project name and there must be no leading or trailing spaces.`

ERROR: test_25_copy_solid_bodies (test_98_Icepak.TestSequenceFunctionsGenerate)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\_unittest_ironpython\conf_unittest.py", line 22, in test
    return getattr(test_obj, test_function)()
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\_unittest\test_98_Icepak.py", line 289, in test_25_copy_solid_bodies
    new_design = Icepak(projectname=project_name, designname=design_name)
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\pyaedt\icepak.py", line 112, in __init__
    FieldAnalysisIcepak.__init__(
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\pyaedt\application\AnalysisIcepak.py", line 73, in __init__
    Analysis.__init__(
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\pyaedt\application\Analysis.py", line 90, in __init__
    Design.__init__(
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\pyaedt\application\Design.py", line 585, in __init__
    self.oproject = project_name
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\pyaedt\generic\general_methods.py", line 255, in inner_function
    return func(*args, **kwargs)
  File "D:\ARM\PyAEDT_UnitTests_IronPython_11242021151639\pyaedt\pyaedt\application\Design.py", line 1115, in oproject
    self._oproject.Rename(os.path.join(self.project_path, proj_name + ".aedt"), True)
Exception: Rename(string pathName, long doOverwrite)(D:ARMPyAEDT_UnitTests_IronPython_11242021151639\IcepakCopiedProject.aedt,True) - call failed. Logs or message window might have more details

----------------------------------------------------------------------

maxcapodi78, 2021-11-26:
try to pass path with / instead of \. I see strange message
The project file path **"D:ARMPyAEDT_UnitTests_IronPython_11242021151639\IcepakCopiedProject.aed.....** showing that the \ have not been recognized.

kbhagat-ansys, 2021-11-30:
> try to pass path with / instead of . I see strange message The project file path **"D:ARMPyAEDT_UnitTests_IronPython_11242021151639\IcepakCopiedProject.aed.....** showing that the \ have not been recognized.

Yes from the description, I am seeing the same issue. Will try "/" and update.

kbhagat-ansys, 2021-11-30:
@maxcapodi78  this is a bug in AEDT's code. Apparently it needs BatchOptions directories to be Forward Slashes and not back slashes. 
With these simple commands one can verify by Launching AEDT and then browsing to Tools > General options > User directories
```batch
set WORKING_DIR=D:\Junk\Core
"C:\Program Files\AnsysEM\v221\Win64\ansysedt.exe" -BatchOptions "tempdirectory='%WORKING_DIR%' Desktop\ProjectDirectory='%WORKING_DIR%' Desktop\PersonalLib='%WORKING_DIR%\PersonalLib' Desktop\userlib='%WORKING_DIR%\userlib'"
```
It works correctly if you do the following 
```batch
set WORKING_DIR=D:/Junk/Core
"C:\Program Files\AnsysEM\v221\Win64\ansysedt.exe" -BatchOptions "tempdirectory='%WORKING_DIR%' Desktop\ProjectDirectory='%WORKING_DIR%' Desktop\PersonalLib='%WORKING_DIR%/PersonalLib' Desktop\userlib='%WORKING_DIR%/userlib'"
```

Thus I think you could close this issue as its a long standing problem.

closed 2021-12-03 by maxcapodi78