ansys/pyaedt #613: pywin32 is not listed in reqs for python 2.7

beliaev-maksim, 2021-12-01:
need to list pywin32 for py2.7
```

ImportError while importing test module 'C:\git\aedt-testing\tests\unittests\test_simulation_data.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
tests\unittests\test_simulation_data.py:9: in <module>
    from aedttest import simulation_data
aedttest\simulation_data.py:31: in <module>
    from pyaedt import get_pyaedt_app  # noqa: E402
venv2_7\lib\site-packages\pyaedt\__init__.py:15: in <module>
    from pyaedt.generic.design_types import Hfss3dLayout
venv2_7\lib\site-packages\pyaedt\generic\design_types.py:17: in <module>
    from pyaedt.hfss3dlayout import Hfss3dLayout
venv2_7\lib\site-packages\pyaedt\hfss3dlayout.py:8: in <module>
    from pyaedt.application.Analysis3DLayout import FieldAnalysis3DLayout
venv2_7\lib\site-packages\pyaedt\application\Analysis3DLayout.py:8: in <module>
    from pyaedt.application.Analysis import Analysis
venv2_7\lib\site-packages\pyaedt\application\Analysis.py:39: in <module>
    from pyaedt.application.Design import Design
venv2_7\lib\site-packages\pyaedt\application\Design.py:25: in <module>
    from pyaedt.desktop import Desktop
venv2_7\lib\site-packages\pyaedt\desktop.py:59: in <module>
    import pythoncom
E   ImportError: No module named pythoncom
```

maxcapodi78, 2021-12-03:
@beliaev-maksim Why we should support CPython 2.7? Is there a specific reason? Otherwise, given that CPython has strong development I would support only from 3.x.

beliaev-maksim, 2021-12-03:
I run a pipeline with Py2.7 instead of IPy. So, standard reqs.txt fails due to missing deps

closed 2021-12-06 by maxcapodi78