ansys/pyaedt #630: Add Support for nonlinear Permeability in Maxwell Materials

DaveTwyman, 2021-12-07:
Request to support nonlinear BH materials through pyaedt.
The below approach (after being tidied up!) can be used to add a nonlinear BH curve

from pyaedt import Maxwell3d

Project_Name = "Add_BH_Curve"
Design_Name = "BH_Curve_Test"
Solver = "Magnetostatic"
DesktopVersion = "2021.2"

M3D = Maxwell3d(projectname=Project_Name,designname=Design_Name,solution_type=Solver,specified_version=DesktopVersion)
uom = M3D.modeler.model_units = "mm"
primitives = M3D.modeler.primitives

def addBH(materialname,B,H):
	BH_range=range(int(0),len(B),int(1))
	List1=[]
	for item in BH_range:
		List1.append(["NAME:Point",H[item],B[item]])
	List1.insert(0,"NAME:BHCoordinates")
	List1.insert(1,["NAME:DimUnits","",""])

	new=M3D.oproject.GetDefinitionManager ()
	new.EditMaterial ("NewMaterial",["NAME:newmaterial","CoordinateSystemType:=","Cartesian","BulkOrSurfaceType:=",1,
		["NAME:PhysicsTypes","set:=",["Electromagnetic","Thermal","Structural"]],["NAME:AttachedData",
			["NAME:MatAppearanceData","property_data:=","appearance_data","Red:=",128,"Green:=",128,"Blue:=",128]],
		["NAME:permeability","property_type:=","nonlinear","BTypeForSingleCurve:=","normal","HUnit:=","A_per_meter",
			"BUnit:=","tesla","IsTemperatureDependent:=",False,List1,
			["NAME:Temperatures"]],"conductivity:=","500000",
		["NAME:magnetic_coercivity","property_type:=","VectorProperty","Magnitude:=","0A_per_meter","DirComp1:=","1",
			"DirComp2:=","0","DirComp3:=","0"]])

MaterialName = 'newmaterial'
B = [0, 0.0025, 0.0050, 0.0125, 0.0250, 0.05, 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9,1.0, 1.1,1.2,1.3,1.4,1.5,1.55,1.6,1.65,1.7,1.75,1.80]
H = [0,16,30,54,93,143,191,210,222,233,247,258,272,289,313,342,377,433,509,648,933,1228,1934,2913,4993,7189,9423]
mat = M3D.materials.add_material(MaterialName)

addBH(MaterialName,B,H)

maxcapodi78, 2021-12-07:
@DaveTwyman Thanks for adding it. Today I received same request from another customer. I'll work on it when I'm back on Thursday

DaveTwyman, 2021-12-07:
@maxcapodi78 , Brilliant thanks, thinking about it being able to import BH curve from a text file would be very useful. The same format as through the Maxwell UI seems to make sense, so '*.bh' and '*.tab'
Let me know if I can help

maxcapodi78, 2021-12-07:
@DaveTwyman right now there is already in pyaedt possibility to export and import materials from json file. It is much more powerful. Give it a try. Once working, bh curves will also be exported and imported

maxcapodi78, 2021-12-10:
#637 Implemented.

DaveTwyman, 2021-12-15:
Works well Thanks

closed 2021-12-15 by DaveTwyman