ansys/pyaedt #655: (AO)Create1Pin Component with PyAEDT

aohta2021, 2021-12-16:
Does anybody have a method to create1Pin Component (not VIA definition) on the board with PyAEDT, or  share a code sample of creating the component?

By AEDT GUI Script can create 1Pin Component, but it is defined just one PARTNAME component.
Since it is not possible to implement by changing multiple REFDES names, it is not efficient because each component is created one partname by one component.
I'm looking for a way to implement multiple REFDES names with  one PARTNAME component.

I would like to use  it for customer's automation service.

Regard.
Akira Ohta

maxcapodi78, 2022-01-11:
@aohta2021  I think it is doable with. The steps are the following:
- create a padstack definifition
- create a PadstackInstance
- create a component based on pin 
- assign a new ComponentDef to the new component
- Reuse the ComponentDef for new single pin components
All the components should have the same Refdes.

Most of the methods are already there. It's just a matter of understanding how it is better to add it.

I was thinking to the following structure: 
`
edbapp.core_padstack.create_padstack(padstackname="myVia")
my_pin = edbapp.core_padstack.place_padstack(position=[0,0], padstackname="myVia", net_name="Mynet")
my_component = edbapp.core_components.create_component_from_pins([my_pin], "newcomp")
`
Now, last command creates 1 ComponentDef (Refdes) for each component with the same name of the component. we can optimize it with a minor modification to the code adding the possibility to "reuse" a componentdef  across multiple components. What do you think?

@ring630 @svandenb-dev Please add your thoughts.

ring630, 2022-03-22:
@maxcapodi78, "Reuse the ComponentDef for new single pin components" sounds risky. The pin number of pins and all relevant properties must be identical. I check AEDT UI, I don't see such command in 3DL. I think the easier way is to 
1, create the 1pin component from padstackinstance.
2, copy and paste the component multiple times.

@svandenb-dev , you know EDB the best. Do we have these command already?
[image]

Samuelopez-ansys, 2022-11-24:
Hi @ring630 , @maxcapodi78,

I think this issue could be closed right?

closed 2023-02-13 by Samuelopez-ansys