ansys/pyaedt #664: Create Radar from JSON with FFD Antenna

Samuelopez-ansys, 2021-12-20:
Unit test and Example shows how to create the JSON file for a parametric antenna, but in the FFD case is not clear how to do it. Adding an example would be interesting.

I have tried to understand how to do it, and I think it should be adding the next parameters to the antennas key in the JSON file:
"antennas": {
		"tx1": {
			"antenna_type":"ffd",
			"ffd_name": "test", 
			"mode":"tx"
}
}

If this syntax is correct, I found a defect in in pyaedt/modeler/parts.py ,line 490, when the ffd path is specified, this variable create a non existing path variable:
ffd = os.path.join(self._compdef["part_folder"], self._multiparts._name + ".ffd", self._compdef["ffd_name"])
>>>ffd = C:\Temp\Example.ffd\test

I think it should be:
ffd = os.path.join(self._compdef["part_folder"], self._compdef["ffd_name"] + ".ffd")
>>>ffd = C:\Temp\test.ffd

But this approach has the limitation that the ffd file must be inside the same folder than the JSON file, so other approaches could be discussed.

maxcapodi78, 2021-12-27:
@dcrawforAtAnsys  can you give a look at it? Or I can if you cannot.

dcrawforAtAnsys, 2021-12-31:
Will try to look next week.

closed 2022-01-11 by maxcapodi78