ansys/pyaedt #668: hfss.post.plot_model_obj() failed to work

linmingchih, 2021-12-21:
the function hfss.post.plot_model_obj() caused an error:

self._plot_from_aedtplt() got an unexpected keyword argument 'jupyter'
Arguments Provided: 
    aedtplt_files = ['D:\\temp\\Model.obj'] 
    imageformat = jpg 
    view = None 
    plot_type = None 
    plot_label = 3D Model 
    model_color = #8faf8f 
    show_model_edge = False 
    off_screen = None 
    scale_min = None 
    scale_max = None 
    jupyter = False

maxcapodi78, 2021-12-27:
@linmingchih I'm refactoring the method right now.
you will have also additional method in object to plot that model only. That will run in non graphical also and requires pyvista to plot it. It works from 2021.R2 which supports obj objects.
Example 
`from pyaedt import Hfss
hfss = Hfss()
box = hfss.modeler.create_box([0,0,0],[10,10,10])
box.plot()`

MaxJPRey, 2021-12-31:
It is now working with the last refactoring made by @maxcapodi78 
I tried the following code:

```
from pyaedt import Hfss
hfss = Hfss()
box = hfss.modeler.create_box([0,0,0],[10,10,10])
box.plot()
box.export_image(r"D:\test.png")
```

_Note: the `export_image()` is calling the `plot_model_obj()` internally_

Thanks again @linmingchih for your feedback.

closed 2021-12-31 by MaxJPRey