ansys/pyaedt #670: Pythonnet is needed to run pyaedt within Linux

amirahm150, 2021-12-21:
Hi, we get the following error, even though pythonnet is already installed, how shall we run pyaedt on our Linux machine directly from the console.

/home/amirahmadi/anaconda3/lib/python3.9/site-packages/pyaedt/generic/process.py:11: UserWarning: Pythonnet is needed to run pyaedt within Linux
  warnings.warn("Pythonnet is needed to run pyaedt within Linux")
/home/amirahmadi/anaconda3/lib/python3.9/site-packages/pyaedt/edb.py:41: UserWarning: Pythonnet is needed to run pyaedt within Linux
  warnings.warn("Pythonnet is needed to run pyaedt within Linux")
/home/amirahmadi/anaconda3/lib/python3.9/site-packages/pyaedt/desktop.py:27: UserWarning: Pythonnet is needed to run pyaedt within Linux
  warnings.warn("Pythonnet is needed to run pyaedt within Linux")
Traceback (most recent call last):
  File "/home/amirahmadi/test.py", line 9, in <module>
    d = Desktop("2021.2", nongraphical)
  File "/home/amirahmadi/anaconda3/lib/python3.9/site-packages/pyaedt/desktop.py", line 282, in __init__
    self._main.interpreter = _com
NameError: name '_com' is not defined

maxcapodi78, 2021-12-27:
@amirahm150 Pyaedt can run on linux in 2 ways only:
- ironpython directly in AEDT
- Cpython using Client Server approach. For last point please refer to https://aedtdocs.pyansys.com/API/ClientServer.html

amrit-poudel, 2021-12-27:
[image]

@maxcapodi78: I keep getting this error message on macosx v 11.6, python v 3.9. I have installed pyaedt v0.4.18 using pip3 install pyaedt ,and installed pythonnet using pip3 install --pre pythonnet=3.0.0a2.

maxcapodi78, 2022-01-11:
@amrit-poudel I don't know what is going on with your installation. I've never tried to install on macosx (only windows and linux). Anyway the problem you are facing is not related to pyaedt but to pythonnet and Mono installation that is needed by pythonnet on OS different from windows.
Pythonnet adds .net framework capabilities to Cpython to run AEDT Api,  but .Net framework is natively supported only by Windows.

MaxJPRey, 2022-01-14:
As @maxcapodi78 mentioned, the problem is related to pythonnet and Mono.

closed 2022-01-14 by MaxJPRey