ansys/pyaedt #679: Cannot instantiate Desktop() object in windows

amrit-poudel, 2021-12-28:
I have installed pyaedt, pythonnet in windows machine with python v 3.7. However, I can't seem to run the following simple command without error:

from pyaedt import Dekstop
d = Desktop(specified_version="2016.1"). 

I do not have any environment variables set and Ansys binary/executable are not in the path. 

I'd appreciate if someone could provide a pointer or two how to instantiate this object and see the Ansys GUI.

I am not sure whether simply specifying "version number" is enough to open the Ansys GUI when using pyaedt. I thought one may also need to specify the path to the Ansys executable, but the Desktop() object does not seem to take the path to the executable as an input.

Thanks for any help!

MaxJPRey, 2021-12-31:
Hi @amrit-poudel,

Based on the error you got, I am assuming that you are working with ANSYS EM Suite 2016.1.
PyAEDT requires ANSYS EM Suite 2021 R1 or later (see https://aedtdocs.pyansys.com/Resources/Installation.html).

I hope it helps you.
Thanks.

amrit-poudel, 2022-01-02:
@MaxJPRey Thanks for that note. The following link says PyAEDT supports prior versions of AEDT, including 2021. 

https://aedtdocs.pyansys.com/

Please see the dependency section. I am a bit confused.

akaszynski, 2022-01-03:
> @MaxJPRey Thanks for that note. The following link says PyAEDT supports prior versions of AEDT, including 2021.

@MaxJPRey, let's update the documentation and provide either a table or list of supported Ansys AEDT releases to clarify this. I agree that it's not clear.

MaxJPRey, 2022-01-13:
The documentation has been improved.
In addition we now provide the user with some information depending on the release version he provides when instantiating a new PyAEDT `Desktop` object.
Work merged in #708 .

closed 2022-01-13 by MaxJPRey