ansys/pyaedt #691: HFSS bondwire abnormal

linmingchih, 2022-01-05:
code can work:
hfss = Hfss(specified_version='2021.2')
hfss.modeler.create_bondwire((0,0,0), (2,2,0), bond_type=1)

code has error:
hfss = Hfss(specified_version='2021.2')
hfss.modeler.create_bondwire((2,2,0), (0,0,0), bond_type=1)

Error Message
-----------------------------------------------------------------------------------------------
Method create_bondwire Failed:  General or AEDT Error. Please Check again
(-2147352567, '發生例外狀況。', (0, None, None, None, 0, -2147024382), None)
pyaedt info: No project is defined. Project Project149 exists and has been read.
pyaedt info: Active design is set to HFSS_RFK
pyaedt info: Design Loaded
pyaedt info: Successfully loaded project materials !
pyaedt info: Materials Loaded
pyaedt error: Method create_bondwire Failed:  General or AEDT Error. Please Check again
pyaedt error: (-2147352567, '發生例外狀況。', (0, None, None, None, 0, -2147024382), None)
Project: Project149, Design: HFSS_RFK (Hybrid Modal Network), [error] Script macro error: Command execution error, please see details in message window. (1:46:21 下午  一月 05, 2022)

Arguments Provided: 
    start_position = (2, 2, 0) 
    end_position = (0, 0, 0) 
    h1 = None 
    h2 = None 
    alpha = None 
    beta = None 
    bond_type = 1 
    diameter = None 
    facets = None 
    name = None 
    matname = None 
Error in :   File "d:\my_venv3\lib\site-packages\pyaedt\modeler\Primitives3D.py", line 461, in create_bondwire
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create_bondwire
pyaedt error: Project: Project149, Design: HFSS_RFK (Hybrid Modal Network), [error] Script macro error: Command execution error, please see details in message window. (1:46:21 下午  一月 05, 2022)

pyaedt error: Arguments Provided: 
pyaedt error:     start_position = (2, 2, 0) 
pyaedt error:     end_position = (0, 0, 0) 
pyaedt error:     h1 = None 
pyaedt error:     h2 = None 
pyaedt error:     alpha = None 
pyaedt error:     beta = None 
pyaedt error:     bond_type = 1 
pyaedt error:     diameter = None 
pyaedt error:     facets = None 
pyaedt error:     name = None 
pyaedt error:     matname = None 
pyaedt error: Error in :   File "d:\my_venv3\lib\site-packages\pyaedt\modeler\Primitives3D.py", line 461, in create_bondwire
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create_bondwire

linmingchih, 2022-01-05:
Primitives3D.py

line 428 is wrong
XSize, YSize, ZSize = self._pos_with_arg(end_position)

It should be 
XSize, YSize, ZSize = self._pos_with_arg([n-m for m, n in zip(start_position, end_position)])

By the way, h2 is start_position[2]-end_position[2], or users need to set h2 to move the height of end_position to the right plane.

MaxJPRey, 2022-01-05:
@linmingchih Thanks for your feedback. I am about to fix it and add new tests according to this modification.

PR #692 (2022-01-06): Improve bond-wire length calculation for each direction.
The current implementation was always assuming that the start position coordinates of the bond-wire were [0, 0, 0] when calculating the length of the bond-wire in each of its directions.

Fix #691 .