ansys/pyaedt #716: How to install PyAEDT into Electronics Desktop in Linux?

frtdys, 2022-01-13:
Hi, I follow the instruction provided by (Title: "Using IronPython in AEDT")
https://aedtdocs.pyansys.com/Resources/Installation.html

i use the following command, try to install the PyAEDT package with embedded ironpython of AEDT.

```
/opt/AnsysEM/AnsysEM21.2/Linux64/common/mono/Linux64/bin/mono
/opt/AnsysEM/AnsysEM21.2/Linux64/common/IronPython/ipy64.exe /home/myname/pyaedt-0.4.19/setup-distutils.py install --user
```

and i got this error

```
error: invalid Python installation: unable to open /opt/AnsysEM/AnsysEM21.2/Linux64/common/IronPython/lib/python2.7/config/Makefile (Could not find a part of the path "/opt/AnsysEM/AnsysEM21.2/Linux64/common/IronPython/lib/python2.7/config/Makefile".)
```

frtdys, 2022-01-16:
Another workaround is to use sys.path.append to add the source code directory of pyaedt, 
then i can import pyaedt in AEDT's ironpython, does it has any difference to the installation method of setup-distutils.py for ironpython?

maxcapodi78, 2022-01-18:
Hello @frtdys this is a good approach. Another simpler approach is to simply extract pyaedt tar.gz into PersonalLib folder of AEDT. 
In this way you don't need to use sys.path.append at all.

Samuelopez-ansys, 2022-04-21:
Hi @frtdys , is this issue resolved? Thanks

MaxJPRey, 2022-07-17:
@Samuelopez-ansys Yes, I agree.

closed 2022-07-17 by MaxJPRey