ansys/pyaedt #740: Changing the magnetization direction of permanent magnets

BLeistr, 2022-01-19:
Which command can be used to change the magnetization direction of, for example, ndfe30 from x to z? 

I also tried it via a json file where I initialized the different directions and then adjust only permeability and coercivity in the code.

m3d.materials.import_materials_from_file('materialien.json')
m3d.materials["ndfe_posz"].update()
m3d.materials["ndfe_posz"].permeability = 1.1
m3d.materials["ndfe_posz"]._update_props("magnetic_coercivity" , -850000)

But by the last line the direction is set to x again.

[materialien.txt](https://github.com/pyansys/pyaedt/files/7898113/materialien.txt)

maxcapodi78, 2022-01-24:
Hello @BLeistr  with latest version this is possible using method set_magnetic_coercitivity. Please look 
at the help https://aedtdocs.pyansys.com/API/_autosummary/pyaedt.modules.Material.Material.set_magnetic_coercitivity.html?highlight=coercitivity#pyaedt.modules.Material.Material.set_magnetic_coercitivity

MaxJPRey, 2022-01-24:
This issue is fixed in version v0.4.23

closed 2022-01-24 by MaxJPRey