ansys/pyaedt #759: 'NoneType' object has no attribute 'GetModule'

amirh92, 2022-01-22:
I intended to run a  given simulation which implemented in Maxwell2d. I encounter 'NoneType' object has no attribute 'GetModule' error from site-packages\pyaedt\application\Analysis.py", line 106.
```python
self._ooptimetrics = self._odesign.GetModule("Optimetrics")
self._ooutput_variable = self._odesign.GetModule("OutputVariable")
```

amirh92, 2022-01-22:
please check this procedure for opening existed simulation. there are a lot of errors that point to the NoneType object having no attribute.

maxcapodi78, 2022-01-24:
@amirh92  can you share a piece of code that reproduce this issue so I can identify the issue?

amirh92, 2022-01-25:
@maxcapodi78 I'm using the last version of pyaedt. thanks for your help.
My project was created with the previous version of aedt. so after opening it with 2022R1 and saving it, my problem was solved.

amirh92, 2022-01-25:
@maxcapodi78 So it works once. after running again the code, I encountered this error
[image]

maxcapodi78, 2022-01-27:
@amirh92 my file is from Examples folder. Can you provide me the file or tell me from which version the project comes?

maxcapodi78, 2022-01-27:
Here the problem is that odesign is not populated and I need to understand why. odesign is created following his steps:
1 I check if a design is present and if is present 
   a. if I passed a design name I check if it exists and if it is of the expected type (maxwell2d in your case)
   b. if a solution type is passed I check if design exists and if it is of expected type
   c. if none of above a new design is created
2. if a design is not present a new design is created
now if you can send me at least the project structure ( a snapshot of project tree is enough) I'll try to reproduce it.

closed 2022-07-14 by Samuelopez-ansys