ansys/pyaedt #774: Add ODB++ Import Function in HFSS 3D Layout

linmingchih, 2022-01-25:
Can't Find ODB++ Import Function in Help

maxcapodi78, 2022-01-25:
Actually you can do with Edb with following method. Once imported, you can save and close the edb and open the edb with Hfss3dLayout.
`
    def import_layout_pcb(self, input_file, working_dir, init_dlls=False, anstranslator_full_path="", use_ppe=False):
`

maxcapodi78, 2022-01-25:
Even easier if to start Edb class directly passing the tgz file (the import layout will be automatically applied.

`
from pyaedt import Edb
edb = Edb("path/to/tgz", edbversion="2021.2")
edb.save_edb()
edb.close_edb()
from pyaedt import Hfss3dLayout
h3d =Hfss3dLayout("path/to/aedb/edb.def")

`

linmingchih, 2022-01-25:
Thanks, It works well.

closed 2022-01-25 by maxcapodi78