ansys/pyaedt #779: Maxwell Circuit Editor not available

JuliusSaitz, 2022-01-26:
For creating complex excitations and winding connections in Maxwell, Maxwell Circuit Editor is typically used. It would be useful, if Maxwell Circuit Editor functions are available in pyaedt,  especially those for creating components (oEditor.CreateComponent), for connecting them (oEditor.CreateWire) and creating the ground (oEditor.CreateGround). Thanks.

maxcapodi78, 2022-01-26:
@JuliusSaitz yes, this is a known limit at the moment. How urgent it is? I'll see if actual circuit class can be reused

JuliusSaitz, 2022-01-26:
@maxcapodi78 One customer actively developing automation using pyaedt has recently asked for it and so I have raised it here. This can still be done using the original functions and so I think it's not urgent.

closed 2022-01-28 by sushobhit-gupt