ansys/pyaedt #783: create_component_from_pins doesn't work on my machine

ring630, 2022-01-26:
Below code is from test_54_create_component_from_pins in test_EDB.py. The unitest passes, but it returns false on my machine. 

pins = self.edbapp.core_components.get_pin_from_component("R13")
print(self.edbapp.core_components.create_component_from_pins(pins, "newcomp"))

result
-----------------------------
(False, None)

MaxJPRey, 2022-01-26:
@ring630 Indeed the test always return a tuple **(False, None)**. Whatever the content of this tuple, the test assertion will always pass.
This tuple **(False, None)** is returned when an exception is raised while calling **PinToConnectable** inside **create_component_from_pins()** .
I am investigating it.
Thanks.

ring630, 2022-01-26:
@MaxJPRey Thanks! Strangly, I modified the code to return False when exception is raised. Unitest failed on my machine, but not on server after pulling.

MaxJPRey, 2022-01-26:
@ring630 The `PinToConnectable()` is now callable from the EdbLib. This final PR #780 should solve this issue.
I will add more assert in the current test, to be sure that the returned tuple is always the correct one.

maxcapodi78, 2022-01-27:
@ring630 Can we close this issue? I assume it is working fine now

ring630, 2022-01-27:
yes it is working fine now. Thanks a lot.

PR #788 (2022-01-28): Improve the test of create_component_from_pins()
Improve the test and its assertions for the component creation.
Make sure that the tuple returned by `create_component_from_pins` is correct.
Fix #783
closed 2022-01-28 by MaxJPRey