ansys/pyaedt #816: Assign Icepak Mesh Region is Failing in 2022 R2

ansys-icepak, 2022-02-03:
Hi, 

This was working fine before, but after I updated AEDT 2022 R2, this method is now failing.   A few snapshots are attached.  Could you please take a look and let me know? 

Thanks,
Best,
Carlos

[image]

[image]

[image]

ansys-icepak, 2022-03-07:
@maxcapodi78 

Hi Massimo, looks like this is still failing in 22 R2.  Have you had a chance to take a look?

maxcapodi78, 2022-05-04:
Hi @ansys-icepak  Have you checked in latest pyaedt version this method? I've added new API settings to be compatible with 22R2 some weeks ago

Samuelopez-ansys, 2022-08-30:
Hi @ansys-icepak ,

Please could you check if the issue is still there? If it works please close the issue.

closed 2022-08-31 by maxcapodi78