ansys/pyaedt #820: Maxwell2d.model_depth(value) gives error:

tizianrot, 2022-02-04:
Hallo,
I tried to use the new function to set the model depth, but whichever syntax variant I tried to use I got the same error:

(<class 'TypeError'>, TypeError('not all arguments converted during string formatting'), <traceback object at 0x000002096B0DE500>)

I ma still using python3.8. May it play a role?
------------
I tried with:
if __name__ == "__main__":
    with Desktop(specified_version=desktopVersion) as desktop:
        M2D = Maxwell2d(solution_type=sType, new_desktop_session=False)
        M2D["Magnetic_Axial_Length"]= "150mm" #works until here

       # I tried them all, one by one
        M2D.model_depth(["Magnetic_Axial_Length"])
        M2D.model_depth("150mm")
        M2D.model_depth("150")
        M2D.model_depth(["150mm"])

thank you,
Tiziana

MaxJPRey, 2022-02-04:
Hi @tizianrot ,

`model_depth` is a property.
Try something like:

```
from pyaedt import Maxwell2d
maxwell_2d = Maxwell2d()
maxwell_2d.solution_type = "Transient"
maxwell_2d.model_depth="150 mm"
```

Regards.

tizianrot, 2022-02-04:
Ah sorry, it was not clear to me since other settings you assign in the same in GUI (i.e.) Matrix computation are achieved via function. As you suggested it works perfectly, thanks and have a great weekend

Tiziana

MaxJPRey, 2022-02-04:
No worries @tizianrot .
Thanks for your feedback. It is truthfully valuable.
Regards,

closed 2022-02-04 by MaxJPRey