ansys/pyaedt #835: Maxwell 3D: How can hidden model objects be included in a png exported by ExportModelImageToFile()?

herbstmichael, 2022-02-08:
Hello Massimo, 
I would like to export a png containing all model objects for documentation purposes. 
The issue:

- `app.modeler.model_objects` contains all model objects including hidden objects
- If the objects  `app.modeler.model_objects` are passed to the `ExportModelImageToFile()` function, the hidden objects are also hidden on the exported png.
- Is there a way to temporarily unhide the objects, export the model image and hide the previously hidden objects again?
- Here is the code I am using for the export:
`param = [
            "NAME:SaveImageParams",
            "ShowAxis:=", "True",
            "ShowGrid:=", "True",
            "ShowRuler:=", "True",
            "ShowRegion:=", "Default",
            "Selections:=", ",".join(app.modeler.model_objects),
            "FieldPlotSelections:=", "",
            "FitToSelections:=", ",".join(app.modeler.model_objects),
            "FitToFieldPlotSelections:=", "",
            "Orientation:=", "view_png_export"
        ]`

`m3d.modeler.oeditor.ExportModelImageToFile(file_name, 0, 0, param)`

Please don't hesitate to contact me for further information. Thanks in advance :-).

maxcapodi78, 2022-02-09:
Hello Michael,
you can use 
m3d.plot()

without any option is simply plotting the full model (inlcuding hidden objects)
with the options:
m3d.plot(show=False, export_path="c:\\temp\image.jpg")
it will not plot but save the picture directly

furthermore you can rung
plt_obj = m3d.plot(show=False)
plt_obj.show_axes = False
...
plt.plot(file_path)

in case you want to customize the plot before generating it.

herbstmichael, 2022-05-30:
Hello akaszynski,
thanks for your quick reply.

Running the command yields the following error:
[image]

Ok, I'll take a look on the link you provided, thank you.
Kind regards 
Michael

closed 2022-08-31 by maxcapodi78