ansys/pyaedt #840: Support '.obj' format in FieldAnalysis3D.export_3d_model()

linmingchih, 2022-02-09:
The .obj format is not supported in FieldAnalysis3D.export_3d_model(). It will be great if you can add it in \pyaedt\application\Analysis3D.py line 488:

        elif fileFormat == '.obj':
            self.modeler.oeditor.ExportModelMeshToFile(os.path.join(filePath, fileName + fileFormat), allObjects)

The reason is that ModelPlotter.add_object() in pyaedt.generic.plot.py seems only support .obj format. Thanks.

linmingchih, 2022-02-10:
Hi Massimo

hfss.plot() can't support orientation setting that ModelPlotter.plot can support. Or you can add orientation in hfss.plot()? Thanks.

Samuelopez-ansys, 2022-08-30:
Hi @linmingchih ,

I am going to close this issue because as Massimo said, with hfss.plot() you could control the orientation and other properties:
[image]

You need to create the plot with 'show' to False, modify the properties and then show it :
my_plot = hfss.plot(show=False, plot_air_objects=False)
my_plot.show_axes = False
my_plot.show_grid = False
my_plot.isometric_view = False
my_plot.camera_position = "xy"
my_plot.off_screen = False
my_plot.plot()

closed 2022-08-30 by Samuelopez-ansys