ansys/pyaedt #846: How to assign the material not in "[sys]Materials" library ?

Siderpan, 2022-02-10:
Hi,

I'm try to assign the material not in ```[sys]Materials``` .
```
m2d.assign_material('Stator', '35CS550')
```
If I just type the name that I want , It well search the whole library and pop out the error 
[image]
Is there something that I misunderstood or wrong type?
And after solving this problem I also want to change the material's parameters , like change ```Composition``` parameter from ```Solid``` to  ```Lamination```.
[image]

Thanks a lot,
Best regards,
Jason.

maxcapodi78, 2022-02-16:
Hi Jason, where is located your material?

maxcapodi78, 2022-02-16:
I've checked and if I've a material saved in an amat file in personallib folder it is correctly retrieved. I've also checked with a granta material and worked as well. Make sure that the material name you add  is correct. In your screenshot "China Steel - 50CS600" and not simply "50CS600"

Siderpan, 2022-02-17:
Hi,
Thanks for your reply. I found that my problem came from using the  old version(2020R2). 
This function works fine when I change the version to 2021R1.

Thank you so much for taking the time to help solve the problem.
Best regards,
Jason

Alberto-DM, 2022-04-01:
As the problem has been solved, I close this issue.

closed 2022-04-01 by Alberto-DM