ansys/pyaedt #847: Improve handling of UDPs

JuliusSaitz, 2022-02-10:
Some suggestions:

1) udp_parameters_list is a required parameter. If none is specified, there is an error. I suggest that this is not required and in case none is specified, the default parameters are used. Currently, if at least one is specified, it works OK.  

2) Remove "name" from the input parameters, as it anyway does not work. The name can be changed later by using UDP.name.

3) Correct the spelling for udp_type parameter. It is written as udptye.

4) Currently, once a UDP is created, there is no way (at least I could not find) to change the UDP parameters. The only way would be use the original API, that is:

oEditor=UDP.m_Editor
oEditor.ChangeProperty(
	[
		"NAME:AllTabs",
		[
			"NAME:Geometry3DCmdTab",
			[
				"NAME:PropServers",
				"rotor:CreateUserDefinedPart:1"
			],
			[
				"NAME:ChangedProps",
				[
					"NAME:InfoCore",
					"Value:="		, "1"
				]
			]
		]
	])

So there is a workaround, but if there is a way to improve this and access the parameters directly, like UDP.InfoCore, that would be great.

MaxJPRey, 2022-02-21:
@JuliusSaitz A new method named `update_udp` has been added.

MaxJPRey, 2022-02-21:
@JuliusSaitz Regarding your feedback from the argument `Name` not having any impact, it should work.
Here is a short example:
```
        my_udpName = "MyClawPoleCore"
        udp = self.aedtapp.modeler.create_udp(
            udp_dll_name="RMxprt/ClawPoleCore",
            udp_parameters_list=my_udpPairs,
            upd_library="syslib",
            name=my_udpName,
            udptye="Solid",
        )

        assert udp
        assert udp.name == "MyClawPoleCore"
```

PR #850 (2022-02-16): Update an existing user-defined primitive (UDP).
It is now possible to update an existing geometry object that was originally created using UDP.
Add unit tests for the UDP updates.
Fix #847 .
closed 2022-02-21 by MaxJPRey