ansys/pyaedt #852: Get data from solved variations during an optimization

IsotropicPaoloS, 2022-02-14:
Hi, while i run an optimization i would like to get data (i.e. s-parameters) from a variation that has already been solved using the ID of that variation (or other way if any) and plot the results outside HFSS(jupyter notebook). 
Is it possible to do this with pyaedt?

I have found this function get_solution_data_per_variation[¶](https://aedtdocs.pyansys.com/API/_autosummary/pyaedt.modules.AdvancedPostProcessing.PostProcessor.get_solution_data_per_variation.html#get-solution-data-per-variation) that seems to do exactly what i want but i'm having some trouble.

First i'm running it on a linux server following this: CPython on Linux with Client-Server[¶](https://aedtdocs.pyansys.com/API/ClientServer.html#cpython-on-linux-with-client-server)

```
from pyaedt.common_rpc import launch_ironpython_server
client = launch_ironpython_server(aedt_path="/datastore/tools/AnsysEM/centos7/AnsysEM21.2/Linux64/", non_graphical=True, port=18000)
```

But once i run the code i get this error

```
Process 53394 started on server1
Using Port 18000
Warning: Remote CPython to Ironpython may have some limitations.
Known Issues are on returned list and dict. 
For those it is recommended to use client.convert_remote_object method.

---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-1-d315426f60ea> in <module>
      4 
      5 from pyaedt.common_rpc import launch_ironpython_server
----> 6 client = launch_ironpython_server(aedt_path="/datastore/tools/AnsysEM/centos7/AnsysEM21.2/Linux64/", non_graphical=True, port=18000)

/datastore/users/psquadrito/hfss_projects/TB4/RX_feed/.venv/lib64/python3.6/site-packages/pyaedt/common_rpc.py in launch_ironpython_server(aedt_path, non_graphical, port, launch_client)
    397     print("For those it is recommended to use client.convert_remote_object method.")
    398     if proc and launch_client:
--> 399         return client(server_name=socket.getfqdn(), server_port=port1)
    400     return False

/datastore/users/psquadrito/hfss_projects/TB4/RX_feed/.venv/lib64/python3.6/site-packages/pyaedt/common_rpc.py in client(server_name, server_port, beta_options)
    236             t -= 1
    237             time.sleep(1)
--> 238     port = c.root.start_service(server_name, beta_options)
    239     if not port:
    240         return "Error Connecting to the Server. Check the server name and port and retry."

UnboundLocalError: local variable 'c' referenced before assignment

```

maxcapodi78, 2022-02-14:
Hello, I don't think it is possible to do at the moment what you are looking for. I need to investigate on it. Anyway what I see on your message above that your server is not launching correctly. This may be due to different reasons. Can you confirm me that you are able to run client-server (when not solving)?

maxcapodi78, 2022-02-16:
are you sure of this path? "/datastore/tools/AnsysEM/centos7/AnsysEM21.2/Linux64/"
Typically there is nothing between AnsysEM and AnsysEM21.2 folders when installing AEDT.

IsotropicPaoloS, 2022-02-16:
Yes the path is correct

Samuelopez-ansys, 2022-03-21:
@maxcapodi78 Could be related to the Python version? I see in the error it is using Python 3.6:
/datastore/users/psquadrito/hfss_projects/TB4/RX_feed/.venv/lib64/**python3.6**/site-packages/pyaedt/common_rpc.py

IsotropicPaoloS, 2022-03-22:
what is the recommended python version?

Samuelopez-ansys, 2022-03-22:
Python 3.7 or 3.8 should be Ok.

Samuelopez-ansys, 2022-11-24:
Hi @AllSpacePsquadrito ,

I will proceed to close this issue. With the new native GRPC native feature, you could use Linux and PyAEDT with no exceptions. This is available from 2022R2.

closed 2022-11-24 by Samuelopez-ansys