ansys/pyaedt #867: Function with several modeler interactions is returning early

cafeclimber, 2022-02-17:
## Description

I have Ansys AEDT running with a project and HFSS design open. I create an Hfss instance as:

```python
with hfss.Hfss() as app:
    # pass app to library
```

and then I pass it to a library I am writing. In that library, I have a function with this signature:

```python
def _draw_patch_antenna(
    hfss: hfss.Hfss, 
    antenna: PatchAntenna, 
    name: Optional[str]
) -> Tuple[Object3d.Object3d, Object3d.Object3d, Object3d.Object3d]
```

that is meant to very simply draw a substrate, rectangular patch, and ground plane.

For some reason in the middle of this function executing the program terminates (from Hfss's perspective it seems). Through debugging, I have been able to determine the line (shown below).

## Expected behavior

My function should run to completion with all calls to the modeler being executed

## Actual behavior

The function returns prematurely with **no** log output indicating why:

```python
def _draw_patch_antenna(
    hfss: hfss.Hfss, antenna: PatchAntenna, name: Optional[str]
) -> Tuple[Object3d.Object3d, Object3d.Object3d, Object3d.Object3d]:
    if name:
        sub_name = f"{name}_substrate"
        patch_name = f"{name}_patch"
    else:
        sub_name, patch_name = None, None

    if not hfss.materials.checkifmaterialexists(antenna.substrate):
        warnings.warn("Failed to add material.")

    modeler = hfss.modeler
    substrate = modeler.create_box(
        position=[-antenna.sub_l / 2, -antenna.sub_w / 2, 0],
        dimensions_list=[antenna.sub_l, antenna.sub_w, antenna.sub_h],
        name=sub_name,
        matname=antenna.substrate
    )
    patch = modeler.create_rectangle(
        csPlane=PLANE.XY,
        position=[-antenna.patch_l / 2, -antenna.patch_w / 2, antenna.sub_h], <====== FAILS HERE =======
        dimension_list=[antenna.patch_l, antenna.patch_w],
        name=patch_name,
    )
    ground = modeler.create_object_from_face(patch.bottom_face_z)
    hfss.create_boundary(boundary_type="PerfectE", sheet_name=[patch.id, ground.id])

    ret = (substrate, patch, ground)
    if not all(ret):
        raise RuntimeError("Failed to create all objects.")

    return ret
```

Here are the logs:

```
2022/02/16 19.45.59:Global:INFO    :Logger Started on C:\Users\<HIDDEN>\Documents\Ansoft\pyaedt20220216_194559.log
2022/02/16 19.45.59:Global:INFO    :pyaedt v0.4.28
2022/02/16 19.45.59:Global:INFO    :Python version 3.9.0 (tags/v3.9.0:9cf6752, Oct  5 2020, 15:34:40) [MSC v.1927 64 bit (AMD64)]
2022/02/16 19.45.59:Global:DEBUG   : 'pyaedt.hfss.Hfss.oproject' has been exectuted.
2022/02/16 19.45.59:Global:DEBUG   :    Implicit Arguments: 'Patch',
2022/02/16 19.45.59:Global:INFO    :AEDT Load time 0.2902193069458008
2022/02/16 19.45.59:Global:DEBUG   : 'pyaedt.hfss.Hfss.odesign' has been exectuted.
2022/02/16 19.45.59:Global:DEBUG   :    Implicit Arguments: 'GapFeedArray',
2022/02/16 19.45.59:Global:INFO    :Design Loaded
2022/02/16 19.45.59:Global:DEBUG   : 'pyaedt.application.design_solutions.HFSSDesignSolution.solution_type' has been exectuted.
2022/02/16 19.45.59:Global:DEBUG   :    Implicit Arguments: None,
2022/02/16 19.45.59:Global:DEBUG   : 'pyaedt.hfss.Hfss.solution_type' has been exectuted.
2022/02/16 19.45.59:Global:DEBUG   :    Implicit Arguments: None,
2022/02/16 19.45.59:Global:INFO    :Successfully loaded project materials !
2022/02/16 19.45.59:Global:INFO    :Materials Loaded
2022/02/16 19.45.59:Global:DEBUG   : 'pyaedt.modeler.Model3D.Modeler3D.refresh_all_ids' has been exectuted.
2022/02/16 19.46.16:Global:DEBUG   : 'pyaedt.modeler.Modeler.CoordinateSystem.update' has been exectuted.
2022/02/16 19.46.16:Global:DEBUG   : 'pyaedt.modeler.Modeler.CoordinateSystem.create' has been exectuted.
2022/02/16 19.46.16:Global:DEBUG   :    Explicit Arguments: 'origin': [-0.4139775228731658, -0.28590724677899976, 1.061001842868719], 'reference_cs': 'Global', 'name': 'Antenna_0', 'mode': 'axis', 'view': 'iso', 'x_pointing': [-0.2883277375759479, -0.29143828221048856, 0.9121024303259666], 'y_pointing': [-0.3065205070664206, -0.21959992175784707, 0.9261862950355569], 'phi': 0, 'theta': 0, 'psi': 0, 'u': None
2022/02/16 19.46.16:Global:DEBUG   : 'pyaedt.modeler.Model3D.Modeler3D.create_coordinate_system' has been exectuted.
2022/02/16 19.46.16:Global:DEBUG   :    Explicit Arguments: 'origin': [-0.4139775228731658, -0.28590724677899976, 1.061001842868719], 'x_pointing': [-0.2883277375759479, -0.29143828221048856, 0.9121024303259666], 'y_pointing': [-0.3065205070664206, -0.21959992175784707, 0.9261862950355569], 'name': 'Antenna_0'
2022/02/16 19.46.20:Global:DEBUG   : 'pyaedt.modeler.Modeler.CoordinateSystem.set_as_working_cs' has been exectuted.
2022/02/16 19.46.40:Global:DEBUG   : 'pyaedt.modules.MaterialLib.Materials.checkifmaterialexists' has been exectuted.
2022/02/16 19.46.40:Global:DEBUG   :    Implicit Arguments: 'Rogers RT/duroid 5880 (tm)',
2022/02/16 19.46.47:Global:DEBUG   : 'pyaedt.modules.MaterialLib.Materials.checkifmaterialexists' has been exectuted.
2022/02/16 19.46.47:Global:DEBUG   :    Implicit Arguments: 'vacuum',
2022/02/16 19.46.47:Global:DEBUG   : 'pyaedt.modules.Material.Material.is_conductor' has been exectuted.
2022/02/16 19.46.47:Global:DEBUG   :    Implicit Arguments: 100000,
2022/02/16 19.46.47:Global:DEBUG   : 'pyaedt.modules.Material.Material.is_dielectric' has been exectuted.
2022/02/16 19.46.47:Global:DEBUG   : 'pyaedt.modeler.Model3D.Modeler3D.create_box' has been exectuted.
2022/02/16 19.46.47:Global:DEBUG   :    Explicit Arguments: 'position': [-0.05, -0.05, 0], 'dimensions_list': [0.1, 0.1, 0.0015748], 'name': 'Antenna_0_substrate'
2022/02/16 19.46.50:Global:DEBUG   : 'pyaedt.modules.MaterialLib.Materials.checkifmaterialexists' has been exectuted.
2022/02/16 19.46.50:Global:DEBUG   :    Implicit Arguments: 'Rogers RT/duroid 5880 (tm)',
2022/02/16 19.46.50:Global:DEBUG   : 'pyaedt.modeler.Model3D.Modeler3D.convert_to_selections' has been exectuted.
2022/02/16 19.46.50:Global:DEBUG   :    Implicit Arguments: 'Antenna_0_substrate', True
```

There is no other output either in my terminal or in the log file.

cafeclimber, 2022-02-17:
After MUCH debugging, it was a typo on my part in the `patch_l` accessor. I wasn't seeing any error output. Is the pyaedt logger hiding stderr or other outputs? While trying to debug, I imported `os` and tried to print some values to `stderr`. Of course, `stderr` is in `sys` not `os`, so I got an error when calling print. But only in one circumstance:

### No error. Early termination

```python
print("Antenna:")
print(f"\tpatch_l: {antenna.patch_l}", file=sys.stderr)
print(f"\tpatch_w: {antenna.patch_w}", file=sys.stderr)
print(f"\tsub_w: {antenna.sub}", file=sys.stderr)
```

### `AttributeError: module os has no attribute stderr`

```python
print("Antenna:")
print(f"\tpatch_w: {antenna.patch_w}", file=sys.stderr)
print(f"\tsub_w: {antenna.sub}", file=sys.stderr)
print(f"\tpatch_l: {antenna.patch_l}", file=sys.stderr)
```

Samuelopez-ansys, 2022-03-21:
Hi @cafeclimber ,

I would need more info to debug the error. I have created the following screenshot where you could play with the different setting setups available: 
[image]

If you don't identify any setting, please, add a piece of code to test the issue. 

Thank you

Samuelopez-ansys, 2022-04-21:
Hi @cafeclimber, did you test my suggestion?

closed 2022-07-14 by Samuelopez-ansys

cafeclimber, 2022-07-27:
Sorry, I haven't had the chance yet. Will update when I'm able!