ansys/pyaedt #875: Get the Same Letter Cases of Material Names Defined in Project

linmingchih, 2022-02-18:
[i.material_name for i in hfss.modeler.object_list] return materials in lower cases even the material name contains upper cases. When I use the name to assign new objects, there will be two categories in the object tree as shown in the figure below. It is confusing.

[image]

maxcapodi78, 2022-02-25:
Hi @linmingchih I'll check into it. The problem is that AEDT is sometimes case insensitive and sometimes case sensitive to material names.

maxcapodi78, 2022-02-25:
@linmingchih this looks a bug since Material Manager is case insensitive. Can you submit a bug to our Dev Team?

Samuelopez-ansys, 2022-05-25:
@maxcapodi78  This is already fixed, right?

closed 2022-11-24 by Samuelopez-ansys