ansys/pyaedt #883: DLL load failed while importing win32api: The specified procedure could not be found

Javpyer, 2022-02-21:
After pyaedt 0.4.31 is installed, pywin32 is updated to pywin32-303, but while open spyder, there's problem occurred: "DLL load failed while importing win32api: The specified procedure could not be found". A change in pywin32 version however will pop error pyaedt 0.4.31 is not compatible w/ older version and spyder won't start. How do I solve this problem? Thanks!

maxcapodi78, 2022-02-25:
Hi @Javpyer have you checked if pywin32 302 is compatible with Spyder? If so I can run some test to se if we can use that version instead of 303

ring630, 2022-07-22:
@maxcapodi78 @Javpyer , I have pywin32 302, and spyder is working fine. I have spyder 5.3.2

MaxJPRey, 2022-08-02:
Hi @Javpyer,
Could you try with the latest version of spyder (5.3.2)?
Thanks.

ring630, 2022-08-18:
@Javpyer , your python environment may be messed up when updating and downgrading packages. Can you create a fresh virtual environment to see if the problem remains?

Samuelopez-ansys, 2023-01-11:
Closing issue. No updates for 6 months.

closed 2023-01-11 by Samuelopez-ansys