ansys/pyaedt #885: Missing module pywintypes

gadiLhv, 2022-02-21:
Using python 3.8, and running "setup.py build" followed by "setup.py install", I am getting this error 
`ModuleNotFoundError: No module named 'pywintypes'`

while running the first import 
`from pyaedt import Desktop, Maxwell3d`

MaxJPRey, 2022-02-21:
Hi @gadiLhv ,

Based on your feedback, I am assuming that you are trying to build pyaedt locally.
Are you using the last version of the main branch?
Which platform are you using to perform those operation?

I tried locally on a Windows machine and it worked.

Thanks for your feedback,

Regards,

Maxime.

gadiLhv, 2022-02-22:
Hi 

I pulled the latest version from the main branch. Then I ran the following two commands using python 3.8.10
python3 setup.py build
python3 setup.py install

I am using pydev in eclipse, and while defining the python3 interpreter, the pyaedt package is clearly visible. Then I try to run the import previously mentioned line, and get the error.

[image]

MaxJPRey, 2022-03-09:
Hi @gadiLhv ,
If you are still facing this issue,
Could you try the following commands in a command prompt (so outside of the eclipse environment to narrow down the issue).
First go the folder containing pyaedt folder pulled from main _(using cd path_to_pyaedt_folder)_
Then enter the commands:
`pip install -e .`
`from pyaedt import Desktop, Maxwell3d`

Regards.

gadiLhv, 2022-03-20:
Same issue

```
pydev debugger: starting (pid: 9040)
Traceback (most recent call last):
  File "c:\repositories\pyaedt\pyaedt\generic\design_types.py", line 5, in <module>
    from pyaedt.hfss3dlayout import Hfss3dLayout
  File "c:\repositories\pyaedt\pyaedt\hfss3dlayout.py", line 8, in <module>
    from pyaedt.application.Analysis3DLayout import FieldAnalysis3DLayout
  File "c:\repositories\pyaedt\pyaedt\application\Analysis3DLayout.py", line 8, in <module>
    from pyaedt.application.Analysis import Analysis
  File "c:\repositories\pyaedt\pyaedt\application\Analysis.py", line 39, in <module>
    from pyaedt.application.Design import Design
  File "c:\repositories\pyaedt\pyaedt\application\Design.py", line 31, in <module>
    from pyaedt.desktop import Desktop
  File "c:\repositories\pyaedt\pyaedt\desktop.py", line 53, in <module>
    import pythoncom
  File "C:\Users\glahav\AppData\Local\Programs\Python\Python38\lib\site-packages\pywin32-303-py3.8-win-amd64.egg\pythoncom.py", line 2, in <module>
    import pywintypes
ModuleNotFoundError: No module named 'pywintypes'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\repositories\pyaedt\pyaedt\__init__.py", line 20, in <module>
    from pyaedt.generic.design_types import Hfss3dLayout
  File "c:\repositories\pyaedt\pyaedt\generic\design_types.py", line 21, in <module>
    from pyaedt.hfss3dlayout import Hfss3dLayout
  File "c:\repositories\pyaedt\pyaedt\hfss3dlayout.py", line 8, in <module>
    from pyaedt.application.Analysis3DLayout import FieldAnalysis3DLayout
  File "c:\repositories\pyaedt\pyaedt\application\Analysis3DLayout.py", line 8, in <module>
    from pyaedt.application.Analysis import Analysis
  File "c:\repositories\pyaedt\pyaedt\application\Analysis.py", line 39, in <module>
    from pyaedt.application.Design import Design
  File "c:\repositories\pyaedt\pyaedt\application\Design.py", line 31, in <module>
    from pyaedt.desktop import Desktop
  File "c:\repositories\pyaedt\pyaedt\desktop.py", line 53, in <module>
    import pythoncom
  File "C:\Users\glahav\AppData\Local\Programs\Python\Python38\lib\site-packages\pywin32-303-py3.8-win-amd64.egg\pythoncom.py", line 2, in <module>
    import pywintypes
ModuleNotFoundError: No module named 'pywintypes'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\repositories\pyaedt\pyaedt\generic\design_types.py", line 5, in <module>
    from pyaedt.hfss3dlayout import Hfss3dLayout
  File "c:\repositories\pyaedt\pyaedt\hfss3dlayout.py", line 8, in <module>
    from pyaedt.application.Analysis3DLayout import FieldAnalysis3DLayout
  File "c:\repositories\pyaedt\pyaedt\application\Analysis3DLayout.py", line 8, in <module>
    from pyaedt.application.Analysis import Analysis
  File "c:\repositories\pyaedt\pyaedt\application\Analysis.py", line 39, in <module>
    from pyaedt.application.Design import Design
  File "c:\repositories\pyaedt\pyaedt\application\Design.py", line 31, in <module>
    from pyaedt.desktop import Desktop
  File "c:\repositories\pyaedt\pyaedt\desktop.py", line 53, in <module>
    import pythoncom
  File "C:\Users\glahav\AppData\Local\Programs\Python\Python38\lib\site-packages\pywin32-303-py3.8-win-amd64.egg\pythoncom.py", line 2, in <module>
    import pywintypes
ModuleNotFoundError: No module named 'pywintypes'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\glahav\.p2\pool\plugins\org.python.pydev.core_9.2.0.202110311311\pysrc\pydevd.py", line 3338, in <module>
    main()
  File "C:\Users\glahav\.p2\pool\plugins\org.python.pydev.core_9.2.0.202110311311\pysrc\pydevd.py", line 3331, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Users\glahav\.p2\pool\plugins\org.python.pydev.core_9.2.0.202110311311\pysrc\pydevd.py", line 2398, in run
    return self._exec(is_module, entry_point_fn, module_name, file, globals, locals)
  File "C:\Users\glahav\.p2\pool\plugins\org.python.pydev.core_9.2.0.202110311311\pysrc\pydevd.py", line 2405, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Users\glahav\.p2\pool\plugins\org.python.pydev.core_9.2.0.202110311311\pysrc\_pydev_imps\_pydev_execfile.py", line 25, in execfile
    exec(compile(contents + "\n", file, 'exec'), glob, loc)
  File "C:\Users\glahav\pyedt-workspace\pyaedt_helloworld\test.py", line 1, in <module>
    from pyaedt import Desktop, Maxwell3d
  File "c:\repositories\pyaedt\pyaedt\__init__.py", line 22, in <module>
    from pyaedt.generic.design_types import Hfss3dLayout
  File "c:\repositories\pyaedt\pyaedt\generic\design_types.py", line 21, in <module>
    from pyaedt.hfss3dlayout import Hfss3dLayout
  File "c:\repositories\pyaedt\pyaedt\hfss3dlayout.py", line 8, in <module>
    from pyaedt.application.Analysis3DLayout import FieldAnalysis3DLayout
  File "c:\repositories\pyaedt\pyaedt\application\Analysis3DLayout.py", line 8, in <module>
    from pyaedt.application.Analysis import Analysis
  File "c:\repositories\pyaedt\pyaedt\application\Analysis.py", line 39, in <module>
    from pyaedt.application.Design import Design
  File "c:\repositories\pyaedt\pyaedt\application\Design.py", line 31, in <module>
    from pyaedt.desktop import Desktop
  File "c:\repositories\pyaedt\pyaedt\desktop.py", line 53, in <module>
    import pythoncom
  File "C:\Users\glahav\AppData\Local\Programs\Python\Python38\lib\site-packages\pywin32-303-py3.8-win-amd64.egg\pythoncom.py", line 2, in <module>
    import pywintypes
ModuleNotFoundError: No module named 'pywintypes'
```

MaxJPRey, 2022-06-13:
Hi @gadiLhv,
Do you still experiment those issues?

Samuelopez-ansys, 2022-09-01:
Hi @gadiLhv ,

Is the issue solved?

Regards

gadiLhv, 2022-09-14:
Hello

No action was really suggested to resolve the issue. I let it go and keep debugging the old way I used to, using the IronPython interpreter with Eclipse.

Samuelopez-ansys, 2022-09-14:
@maxcapodi78 Any suggestion to help @gadiLhv ?

closed 2023-02-13 by Samuelopez-ansys