ansys/pyaedt #915: EdbStackup.flip_design does not mirror primitives, which is expected by EdbStackup.place_in_layout

isaacansys, 2022-03-03:
[EdbStackup.flip_design](https://github.com/pyansys/pyaedt/blob/e44aab5c24dd25fe1b5bc8776af121b4517f48ba/pyaedt/edb_core/stackup.py#L552) leaves all primitives as-is.  This results in incorrect behavior of [EdbStackup.place_in_layout](https://github.com/pyansys/pyaedt/blob/e44aab5c24dd25fe1b5bc8776af121b4517f48ba/pyaedt/edb_core/stackup.py#L338) when the place_on_top argument is False.  In this case flip_design is called on the host design; the stackup is reversed but the primitives (and other items like padstack pad shapes) are never mirrored because there is no way to set the "transform" of the host design.

I have pushed a branch with a [test](https://github.com/pyansys/pyaedt/blob/e26714b5aa580286afdd750f0e131acbd86712d4/_unittest/test_00_EDB.py#L764) for flip_design that checks whether primitives are flipped, which currently fails.

ring630, 2022-03-30:
For a complete flip, not only the primitives itself needs to be flipped but also its xy location as well as traces, shapes, etc.

svandenb-dev, 2022-04-13:
@isaacansys @ring630 what is the status ? is it still needed since we have the 3D placement implemented now ?

svandenb-dev, 2022-07-05:
@isaacansys,
I agree with renaming the the command to "invert_stackup" with explaining in docstring that only the stakup is revert

isaacansys, 2022-10-11:
Rename and documentation looks good to me.  Will also have to remove EdbStackup.place_in_layout which relies on this method to actually flip the design, leaving only place_in_layout_3d_placement for now.

[External Sender]

Samuelopez-ansys, 2023-02-13:
@svandenb-dev @ring630 Could we close this issue?

closed 2023-02-13 by Samuelopez-ansys