ansys/pyaedt #949: hfss create port from selected edges

boyang2022, 2022-03-14:
hfss create port from selected edges

MaxJPRey, 2022-03-16:
Hi @boyang2022 
The following lines of code allow you to create some edges and then, select them and create a circuit port from them.

```
from pyaedt import Hfss
hfss_app = Hfss()
plane = hfss_app.PLANE.XY
rectangle1 = hfss_app.modeler.create_rectangle(plane, [10, 10, 10], [10, 10], name="rectangle1_for_port")
edge1 = hfss_app.modeler.get_object_edges(rectangle1.id)[0]

rectangle2 = hfss_app.modeler.create_rectangle(plane, [30, 10, 10], [10, 10], name="rectangle2_for_port")
edge2 = hfss_app.modeler.get_object_edges(rectangle2.id)[0]

hfss_app.create_circuit_port_from_edges(edge1, edge2, port_name="my_port", port_impedance=50.1, renormalize=False, renorm_impedance="50+1i*55")
```
Could this solve your issue?

Best Regards,

Maxime

boyang2022, 2022-03-17:
hello @MaxJPRey , 

Need lumped port in this case. Basically, selected two edges, create the sheet then create the port. 

Now it is doable in hffs by just select two edges.

@Samuelopez-ansys 
/Bo

Samuelopez-ansys, 2022-03-22:
Hi @boyang2022 ,

I am working on this feature. And I have already implemented it in the branch: fix_issue949. But I detected an issue about how the lumped port is created.

@Alberto-DM  I think you worked on this part. In the lumped port creation, the impedance is not specified, it is fixed to 50 ohms, is it OK? I think it should be mutable.

Please could you check the branch? I think this could be improved, for example, method "create_lumped_port_between_objects", only takes the impedance for Modal solution type, but for the terminal, how to define the renormalization impedance?

boyang2022, 2022-03-23:
@Samuelopez-ansys  thanks for the update.

Alberto-DM, 2022-04-27:
@Samuelopez-ansys impedance support for Lumped ports and Wave ports in Terminal mode is done in PR #1103

closed 2022-04-27 by Alberto-DM