ansys/pyaedt #952: modeler insert segmented Spline

boyang2022, 2022-03-14:
tried the following: 

```python
from pyaedt.desktop import Desktop
from pyaedt import Q3d
from pyaedt.modeler.Primitives import PolylineSegment

d = Desktop("2022.1", False, False)
q3d = Q3d(designname="test", specified_version="2022.1")
q3d.modeler.model_units = "mm"
q3d.modeler.delete()

point = [[1., 1., 0], [1., 2., 1.], [1., 3., 1.]]

line1 = q3d.modeler.create_polyline([[0, 0, 0], point[0]])

line1.insert_segment(position_list=point,
                     segment=PolylineSegment("Spline", num_points=3, num_seg=3))

```
And insert part is not segmented.
[image]

Need to insert segmented Spline.

in PolylineSegment spine has no num_seg
[image]

MaxJPRey, 2022-03-15:
Hi @boyang2022 ,

Following is an example that should work and create the spline as you are expecting it to be.

```
from pyaedt.desktop import Desktop
from pyaedt import Q3d
from pyaedt.modeler.Primitives import PolylineSegment

d = Desktop("2022.1", False, False)
q3d = Q3d(designname="test", specified_version="2022.1")
q3d.modeler.model_units = "mm"
q3d.modeler.delete()

point = [[1., 1., 0], [1., 2., 1.], [1., 3., 1.]]

q3d.modeler.create_polyline(position_list=point, segment_type=PolylineSegment("Spline", num_points=3, num_seg=3))
```

Please let me know if that solves the issue you faced.

Best Regards.

boyang2022, 2022-03-16:
hello @MaxJPRey thanks for your reply.

Yes, I can create a spline. However, I want is to **insert** a segmented spline after a straight line. For example,
[image]

MaxJPRey, 2022-03-27:
Hi @boyang2022 ,
This will be corrected in the next release (v0.4.46).
Thanks.

PR #986 (2022-03-28): Split a polyline with a certain number of segment.
Add the ability to split a polyline with a certain number of segment.
Fix #952 .
Fix #975 .
closed 2022-03-28 by maxcapodi78