ansys/pyaedt #1025: [Maxwell2d] symmetry multiplier

mehs1006, 2022-04-04:
Hi,

I tried to use the module of "symmetry_multiplier" in PyAEDT to make the below design setting.
[image]

But it seems this function doesn't work for resetting "Multiplier" to another value.

Please guide me on how to set the "symmetry_multiplier".
Thanks a lot.

BR,
/Meng-Ju

maxcapodi78, 2022-04-15:
Hello @mehs1006 I've added 2 methods in #1044 
app.change_design_settings({"Multiplier": 2})
app.change_simmetry_multiplier(1)
The first one is generic and can be used to chage any design settings by passing as a dictionary. The second is specific for simmetry multiplier

closed 2022-04-15 by maxcapodi78

mehs1006, 2022-04-19:
Hi @maxcapodi78 ,
The first one is even better and more flexible!
Thank you. 👍