ansys/pyaedt #1026: 3D component creation is not reading Terminal ports properly

mahesh-dhanekula, 2022-04-05:
I am using create_3dcomponent() of pyaedt to create a 3d component. 
I observed there is a difference in 3d component created between pyaedt method and manual method (selected everything in modeler to create 3dcomponent).
Solution type of the design is Terminal Network.
Difference I have observed is with respect to way the Excitations are read in both cases. I have used the script recording feature to capture the following information.
In Manual Method, I am getting Excitation list as **["LISN_P","Load_2_T1","Load_1_T1","LISN__P","Load_2","Load_1"]**
In PyAEDT Method, I am getting Excitation list as **["LISN_P","Load_2_T1","Load_1_T1"]**
Due to this I am getting following errors while validation 

    [error] Terminal'LISN_P': Terminals must be assigned to conductor faces or edges that touch a port.  Please create one or more ports before assigning terminals
    [error] Terminal'Load_2_T1': Terminals must be assigned to conductor faces or edges that touch a port.  Please create one or more ports before assigning terminals
    [error] Terminal'Load_1_T1': Terminals must be assigned to conductor faces or edges that touch a port.  Please create one or more ports before assigning terminals

To reproduce this issue, use the below code:

`import os, random, tempfile, shutil
from pyaedt import Hfss
path_3dcmp = "C:/Program Files/AnsysEM/v221/Win64/syslib/3DComponents/HFSS/EMI EMC/Radiated Emissions/RE_CISPR25_4_cables.a3dcomp"
tmp_folder = tempfile.mkdtemp(prefix="RE_Test_")
shutil.copy(path_3dcmp, tmp_folder)
tmp_3dcomp = tmp_folder+"/RE_CISPR25_4_cables.a3dcomp"
h = Hfss(tmp_3dcomp)
h.save_project(tmp_folder+"/Test3d.aedt")
h.modeler.create_3dcomponent(os.path.join(tmp_folder,"Testing.a3dcomp"))
h.close_project()
h = Hfss(os.path.join(tmp_folder,"Testing.a3dcomp"))
h.release_desktop(False, False)`

**Manual Method**
`oEditor.Create3DComponent(
	[
		"NAME:CreateData",
		"ComponentName:="	, "RE_CISPR25_4_cables",
		"Company:="		, "",
		"Company URL:="		, "",
		"Model Number:="	, "",
		"Help URL:="		, "",
		"Version:="		, "1.0",
		"Notes:="		, "",
		"IconType:="		, "",
		"Owner:="		, "Mahesh Babu Dhanekula",
		"Email:="		, "",
		"Date:="		, "10:43:56 AM  Apr 05, 2022",
		"HasLabel:="		, False,
		"IsEncrypted:="		, False,
		"AllowEdit:="		, False,
		"SecurityMessage:="	, "",
		"Password:="		, "",
		"EditPassword:="	, "",
		"PasswordType:="	, "UnknownPassword",
		"HideContents:="	, True,
		"ReplaceNames:="	, True,
		"ComponentOutline:="	, "None",
		"IncludedParts:="	, ["Support_EUT","Table_Reference_Plane","Conductor","Insulator","Conductor_1","Conductor_2","Conductor_3","Insulator_1","Insulator_2","Insulator_3","Load_2","Load","Load_1","Load_2_1","LISN","Support_Cable","Wood_Table","Chamber_Floor","Chamber_Walls","EUT_Bounding_Box","Antenna_Position_1m"],
		"HiddenParts:="		, [],
		"IncludedCS:="		, ["EUT"],
		"ReferenceCS:="		, "EUT",
		"IncludedParameters:="	, ["Antenna_Distance_From_Cable","Antenna_Height_From_Table_Reference_Plane","Cable_Conductor_Diameter","Cable_Distance_From_Edge","Cable_Facets","Cable_Insulator_Diameter","Cable_Length","Cable_Offset","Cable_Separation","Chamber_Height","Chamber_Wall_Back_Position","Chamber_Wall_Front_Position","Chamber_Wall_Lateral_Postion","EUT_Bounding_Box","Support_Cable_Width","Support_EUT_Length","Support_EUT_Width","Support_Height","Table_Height","Table_Reference_Plane_Back_Position","Table_Reference_Plane_Front_Position"],
		"IncludedDependentParameters:=", [],
		"ParameterDescription:=", [			"Antenna_Distance_From_Cable:=", "",			"Antenna_Height_From_Table_Reference_Plane:=", "",			"Cable_Conductor_Diameter:=", "",			"Cable_Distance_From_Edge:=", "",			"Cable_Facets:="	, "",			"Cable_Insulator_Diameter:=", "",			"Cable_Length:="	, "",			"Cable_Offset:="	, "",			"Cable_Separation:="	, "",			"Chamber_Height:="	, "",			"Chamber_Wall_Back_Position:=", "",			"Chamber_Wall_Front_Position:=", "",			"Chamber_Wall_Lateral_Postion:=", "",			"EUT_Bounding_Box:="	, "",			"Support_Cable_Width:="	, "",			"Support_EUT_Length:="	, "",			"Support_EUT_Width:="	, "",			"Support_Height:="	, "",			"Table_Height:="	, "",			"Table_Reference_Plane_Back_Position:=", "",			"Table_Reference_Plane_Front_Position:=", ""],
		"IsLicensed:="		, False,
		"LicensingDllName:="	, "",
		"VendorComponentIdentifier:=", "",
		"PublicKeyFile:="	, ""
	], 
	[
		"NAME:DesignData",
		"Boundaries:="		, ["Table_Reference_Plane_Table","GND_Cable_LISN","Chamber_Floor","Rad1"],
		"Excitations:="		, ["LISN_P","Load_2_T1","Load_1_T1","LISN__P","Load_2","Load_1"]
	], "D:/Project/AEMC/3Dcomponent/Testing.a3dcomp", 
	[
		"NAME:ImageFile",
		"ImageFile:="		, ""
	])`

**PyAEDT Method**
`oEditor.Create3DComponent(
	[
		"NAME:CreateData",
		"ComponentName:="	, "RE_CISPR25_4_cables",
		"Company:="		, "",
		"Company URL:="		, "",
		"Model Number:="	, "",
		"Help URL:="		, "",
		"Version:="		, "1.0",
		"Notes:="		, "",
		"IconType:="		, "",
		"Owner:="		, "pyaedt",
		"Email:="		, "",
		"Date:="		, "9:44:15 AM  Mar 03, 2021",
		"HasLabel:="		, False,
		"IsEncrypted:="		, False,
		"AllowEdit:="		, False,
		"SecurityMessage:="	, "",
		"Password:="		, "",
		"EditPassword:="	, "",
		"PasswordType:="	, "UnknownPassword",
		"HideContents:="	, True,
		"ReplaceNames:="	, True,
		"ComponentOutline:="	, "None",
		"IncludedParts:="	, ["Support_EUT","Conductor","Insulator","Conductor_1","Conductor_2","Conductor_3","Insulator_1","Insulator_2","Insulator_3","LISN","Support_Cable","Wood_Table","Chamber_Walls","Table_Reference_Plane","Load_2","Load","Load_1","Load_2_1","Chamber_Floor","EUT_Bounding_Box","Antenna_Position_1m"],
		"HiddenParts:="		, [],
		"IncludedCS:="		, ["Global","EUT","Antenna_Location_1m"],
		"ReferenceCS:="		, "EUT",
		"IncludedParameters:="	, [],
		"IncludedDependentParameters:=", [],
		"ParameterDescription:=", [],
		"IsLicensed:="		, False,
		"LicensingDllName:="	, "",
		"VendorComponentIdentifier:=", "",
		"PublicKeyFile:="	, ""
	], 
	[
		"NAME:DesignData",
		"Boundaries:="		, ["Table_Reference_Plane_Table","GND_Cable_LISN","Chamber_Floor","Rad1"],
		"Excitations:="		, ["LISN_P","Load_2_T1","Load_1_T1"]
	], "D:/Project/AEMC/3Dcomponent/Testing.a3dcomp", 
	[
		"NAME:ImageFile",
		"ImageFile:="		, ""
	])`

maxcapodi78, 2022-04-14:
@mahesh-dhanekula I've fixed in #1041 . It seems that when is a terminal solution the script expect both the port names and the terminal names

mahesh-dhanekula, 2022-04-14:
@maxcapodi78 Can you also check in case of modal solution as well ? Because I am not able to see excitations argument getting parsed in 3D component created through this method for modal case

**Manual Method**

```
oDesktop.OpenProject("Magic_Tee.a3dcomp")
oEditor.Create3DComponent(
	[
		"NAME:CreateData",
		"ComponentName:="	, "MagicTee",
		"Company:="		, "ANSYS",
		"Company URL:="		, "",
		"Model Number:="	, "",
		"Help URL:="		, "",
		"Version:="		, "1.0",
		"Notes:="		, "",
		"IconType:="		, "",
		"Owner:="		, "Mahesh Babu Dhanekula",
		"Email:="		, "maheshbabu.dhanekula@ansys.com",
		"Date:="		, "12:50:21 PM  Apr 08, 2022",
		"HasLabel:="		, False,
		"IsEncrypted:="		, False,
		"AllowEdit:="		, False,
		"SecurityMessage:="	, "",
		"Password:="		, "",
		"EditPassword:="	, "",
		"PasswordType:="	, "UnknownPassword",
		"HideContents:="	, True,
		"ReplaceNames:="	, True,
		"ComponentOutline:="	, "None",
		"IncludedParts:="	, ["MT_Interior","MT"],
		"HiddenParts:="		, [],
		"IncludedCS:="		, ["CenterCS","aEndCS","bEndCS","cEndCS"],
		"ReferenceCS:="		, "Global",
		"IncludedParameters:="	, ["Flange","FlangeFillet","FlangeThickness","GuideHeight","GuideLength","GuideWidth","WallThickness"],
		"IncludedDependentParameters:=", [],
		"ParameterDescription:=", [			"Flange:="		, "",			"FlangeFillet:="	, "",			"FlangeThickness:="	, "",			"GuideHeight:="		, "",			"GuideLength:="		, "",			"GuideWidth:="		, "",			"WallThickness:="	, ""],
		"IsLicensed:="		, False,
		"LicensingDllName:="	, "",
		"VendorComponentIdentifier:=", "",
		"PublicKeyFile:="	, ""
	], 
	[
		"NAME:DesignData",
		"Excitations:="		, ["1","2","3","4"]
	], "D:/Project/AEMC/3Dcomponent/TestingWD.a3dcomp", 
	[
		"NAME:ImageFile",
		"ImageFile:="		, ""
	])
```

**PyAEDT Method**
```
oDesktop.OpenProject("Magic_Tee.a3dcomp")
oEditor.Create3DComponent(
	[
		"NAME:CreateData",
		"ComponentName:="	, "MagicTee",
		"Company:="		, "",
		"Company URL:="		, "",
		"Model Number:="	, "",
		"Help URL:="		, "",
		"Version:="		, "1.0",
		"Notes:="		, "",
		"IconType:="		, "",
		"Owner:="		, "pyaedt",
		"Email:="		, "",
		"Date:="		, "9:44:15 AM  Mar 03, 2021",
		"HasLabel:="		, False,
		"IsEncrypted:="		, False,
		"AllowEdit:="		, False,
		"SecurityMessage:="	, "",
		"Password:="		, "",
		"EditPassword:="	, "",
		"PasswordType:="	, "UnknownPassword",
		"HideContents:="	, True,
		"ReplaceNames:="	, True,
		"ComponentOutline:="	, "None",
		"IncludedParts:="	, ["MT_Interior","MT"],
		"HiddenParts:="		, [],
		"IncludedCS:="		, ["Global","CenterCS","aEndCS","bEndCS","cEndCS"],
		"ReferenceCS:="		, "Global",
		"IncludedParameters:="	, [],
		"IncludedDependentParameters:=", [],
		"ParameterDescription:=", [],
		"IsLicensed:="		, False,
		"LicensingDllName:="	, "",
		"VendorComponentIdentifier:=", "",
		"PublicKeyFile:="	, ""
	], 
	[
		"NAME:DesignData"
	], "c:/users/mdhaneku/appdata/local/temp/RE_Test_w45zj9/Testing.a3dcomp", 
	[
		"NAME:ImageFile",
		"ImageFile:="		, ""
	])
```

maxcapodi78, 2022-04-15:
Done in #1044

closed 2022-04-15 by maxcapodi78